ros-infrastructure/rosdep #497: Installing pip packages from setup.py `requires` section?

peci1, 2017-01-26:
What is the `requires` section in setup.py file good for when rosdep ignores it? I'd expect that when I add something as a dependency there, rosdep would automatically try to pip-install it, which, however, doesn't happen.

I'd like to start a discussion about this, because this section is really confusing for a lot of users ( e.g. http://answers.ros.org/question/229250/catkin_make-not-installing-python-dependencies/ ).

One of the problems may be that there is not a word about this section in [documentation](http://docs.ros.org/indigo/api/catkin/html/user_guide/setup_dot_py.html), but it is normally used throughout the [ROS codebase](https://github.com/ros/geometry2/blob/indigo-devel/tf2_ros/setup.py).

If the only meaning of the `requires` section is generating metadata for PyPi, I think it should really be mentioned somewhere.

On the other hand, I'd consider it a nice option if rosdep could automatically use pip to install these dependencies (though it'd be the third file where you can seek for dependencies, which could be unfortunate).

dirk-thomas, 2017-01-26:
In ROS packages the dependencies are declared in the manifest file `package.xml`. You can read more about this in the [REP 140](http://www.ros.org/reps/rep-0140.html). ROS packages even when they contain a `setup.py` file can't be released via PyPi since the Python part is only part of a ROS package.

`rosdep` will iterate over all dependencies declared in the manifest and install those using the mapping provided in https://github.com/ros/rosdistro/tree/master/rosdep If they map to a `pip` installed package rosdep will use that. If you want to package a ROS package for Debian it can't depend on packages not available as Debian package (so depending on a rosdep key which maps to `pip` will prevent the ROS package to be packaged into a Debian package).

peci1, 2017-01-26:
Thanks for the answer.

So does it make sense to explicitly state it on the wiki/in the documentation, that the requirements field has no meaning in ROS?

If I understand it correctly, even the tf2_ros/setup.py has these unnecessary requirements. At least from what I see around me, people often get inspired by looking into some core ROS packages. Would it make sense to start an initiative to clean up these files at least for packages in the ros github organization?

dirk-thomas, 2017-01-26:
Yes, those fields are not being used in ROS packages. It would be great if you could update the wiki page(s). If you like you can post a link to the diff here for others to find / comment on the change.

peci1, 2017-01-26:
I've added a link to setup.py docs to http://wiki.ros.org/catkin , which has IMO it's place there.

And what about this change to http://wiki.ros.org/catkin/CMakeLists.txt ?

    12a13
    >  1. '''Enable Python module support''' ({{{catkin_python_setup()}}})
    248d248
    < 
    254d253
    < 
    260d258
    < 
    303a302,315
    > 
    > == Enabling Python module support ==
    > If your ROS package provides some Python scripts or modules, you should create a [[http://docs.ros.org/api/catkin/html/user_guide/setup_dot_py.html|setup.py]] file and call
    > 
    > {{{
    > catkin_python_setup()
    > }}}
    > before the call to {{{generate_messages()}}} and {{{catkin_package()}}}.
    > 
    > If you want to install a Python script, call the following command:
    > 
    > {{{
    > catkin_install_python(PROGRAMS scripts/myscript DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION})
    > }}}
    344d355
    < 
    345a357
    >

dirk-thomas, 2017-01-26:
The first part looks good. The second part is already covered on the wiki page: http://wiki.ros.org/catkin/CMakeLists.txt#Installing_Python_Executable_Scripts

peci1, 2017-01-26:
Ah, missed that one... I'll remove it. One thing is unclear to me (and maybe it'd be good to state it on the wiki): do I need to call catkin_python_setup and create setup.py if I only want to call catkin_install_python?

dirk-thomas, 2017-01-26:
No, you can call `catkin_install_python` independently. It only installs the scripts to the libexec folder where it can be found by `rosrun`. The `setup.py`  / `catkin_python_setup` takes care of installing the Pytohn modules.

peci1, 2017-01-26:
And what about adding e.g. a `--python-modules` option to `catkin_create_pkg`, which would also create a correct setup.py and add the catkin_python_setup call to CMakeLists?

dirk-thomas, 2017-01-26:
I never use that script so can't really tell if it would be helpful. If you think it would be helpful please consider making a PR for it.

peci1, 2017-01-26:
I find it quite helpful :) Let's see if I'll find time for it...

Otherwise, the proposed wiki change looks good to you?

peci1, 2017-01-27:
Wiki is edited. Thanks for comments ;)

closed 2017-01-27 by peci1