ros-infrastructure/rosdep #500: install only the top-level dependencies of a package?

samlelouch, 2017-03-01:
I am working on Ubuntu 16.04 ARM and looking to install ROS packages. However, besides rosdep, is there an `apt-get install` command or any other command that will let me install only the package's top-level dependencies and not all of them recursively?

wjwwood, 2017-03-02:
Besides `rosdep`? No, not that I'm aware of. `rosinstall_generator` is able to control the "depth" of the dependencies, but it just clones source code. It doesn't install things from the package manager.

wjwwood, 2017-03-02:
Also, you might get better or faster feedback on answers.ros.org, but you'll probably want to include more information about what you're trying to accomplish. Since this isn't a bug or feature request of `rosdep` (yet), I'm going to close this in favor of asking on answers.ros.org, see also: http://wiki.ros.org/Support

closed 2017-03-02 by wjwwood