ros-infrastructure/rosdep #501: rosdep install fails on OSX with error homebrew: Failed to detect successful installation of [freetype]

Maryam81609, 2017-03-02:
When I run the following command:
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y

It fails with the warning:
Warning: freetype-2.7.1 already installed, it's just not linked.

and the error:
homebrew: Failed to detect successful installation of [freetype]

Maryam81609, 2017-03-03:
I solved this by uninstalling and installing freetype.

wjwwood, 2017-03-03:
Yeah, `rosdep` requires that the installed thing be linked by Homebrew to `/usr/local` before considering it "installed". You could have also done `brew link freetype`. Sorry I didn't get back to you before you figured it out on your own.

In the future you might get faster responses from answers.ros.org.

closed 2017-03-03 by wjwwood