ros-infrastructure/rosdep #502: Missing C++ compiler dependency

s-moffett, 2017-03-03:
New to ROS here, please excuse the dumb questions. I'm not even sure if this issue belongs here.

ROS Kinetic on Ubuntu 16.04 in VirtualBox.

On my first `catkin_make` I got errors about missing a C++ compiler. Maybe g++ or build-essential should be standard dependencies?

I did the standard `apt-get install ros-kinetic-desktop-full` then `rosdep init` and `rosdep update` and finally `apt-get install python-rosinstall`.

Attached are the CMake logs and my apt log and a copy-paste of the catkin-make terminal output.

[ROS-apt-log.txt](https://github.com/ros-infrastructure/rosdep/files/817552/ROS-apt-log.txt)
[ROS-make-log.txt](https://github.com/ros-infrastructure/rosdep/files/817553/ROS-make-log.txt)

[CMakeError.txt](https://github.com/ros-infrastructure/rosdep/files/817556/CMakeError.txt)
[CMakeOutput.txt](https://github.com/ros-infrastructure/rosdep/files/817557/CMakeOutput.txt)

dirk-thomas, 2017-03-03:
If you follow the [from-binary install instructions](http://wiki.ros.org/kinetic/Installation/Ubuntu) they only mention what you need to *run* ROS. If you want to build packages yourself you might want to look at the [from-source instructions](http://wiki.ros.org/kinetic/Installation/Source#Installing_bootstrap_dependencies) which mention explicitly which dependencies you need to install (including `build-essential`.

I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) in the future.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you need further help please asked your question on answers.ros.org and feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2017-03-03 by dirk-thomas

s-moffett, 2017-03-03:
Thanks for the kind response, I see exactly what you mean.

Maybe a note should be added in the from-binary install instructions for the beginners who are following the Installing and Configuring tutorial. I'll follow up with answers.ros.org.

dirk-thomas, 2017-03-03:
That might indeed be helpful. You can either edit the wiki page directly or if you are unsure about the change you can propose the diff in a ticket in the [roswiki](https://github.com/ros-infrastructure/roswiki) repo and get feedback before actually changing it.

mikaelarguedas, 2017-03-06:
moved to https://github.com/ros-infrastructure/roswiki/issues/201