ros-infrastructure/rosdep #503: ERROR: Rosdep experienced an error: rosdep data for [vcs_uri] must be a dictionary

braintjy, 2017-03-18:
I followed the instructions from [raspicam_node from UbiquityRobotics](https://github.com/UbiquityRobotics/raspicam_node) to install raspicam_node on my Raspberry PI 3 with a v1.3 camera. I'm using ROS Kinetic. I'm currently stuck at this command rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y which gives an error shown below. Can anyone help? Thanks!

> ERROR: Rosdep experienced an error: rosdep data for [vcs_uri] must be a dictionary
> Please go to the rosdep page [1] and file a bug report with the stack trace below.
> [1] : http://www.ros.org/wiki/rosdep
> 
> rosdep version: 0.11.5
> 
> Traceback (most recent call last):
>   File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
>     exit_code = _rosdep_main(args)
>   File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 353, in _rosdep_main
>     return _package_args_handler(command, parser, options, args)
>   File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 447, in _package_args_handler
>     return command_handlers[command](lookup, packages, options)
>   File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 631, in command_install
>     uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
>   File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 409, in get_uninstalled
>     resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
>   File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 383, in resolve_all
>     rosdep_keys = self.get_rosdeps(resource_name, implicit=implicit)
>   File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 305, in get_rosdeps
>     return self.loader.get_rosdeps(resource_name, implicit=implicit)
>   File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 127, in get_rosdeps
>     m = self._rospack.get_manifest(resource_name)
>   File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 167, in get_manifest
>     return self._load_manifest(name)
>   File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 211, in _load_manifest
>     retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
>   File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 412, in parse_manifest_file
>     _static_rosdep_view = init_rospack_interface()
>   File "/usr/lib/python2.7/dist-packages/rosdep2/rospack.py", line 59, in init_rospack_interface
>     return lookup.get_rosdep_view(DEFAULT_VIEW_KEY)
>   File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 605, in get_rosdep_view
>     view = self.create_rosdep_view(view_key, dependencies + [view_key], verbose=verbose)
>   File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 550, in create_rosdep_view
>     view.merge(db_entry, verbose=verbose)
>   File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 211, in merge
>     update_definition = RosdepDefinition(dep_name, dep_data, update_entry.origin)
>   File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 65, in __init__
>     raise InvalidData("rosdep data for [%s] must be a dictionary"  % (self.rosdep_key), origin=origin)
> InvalidData: rosdep data for [vcs_uri] must be a dictionary

tfoote, 2017-03-20:
Unfortunately this is a misleading error. It's not a problem with rosdep, but that it's misconfigured with an incorrect url. I'd suggest filing a ticket against https://github.com/UbiquityRobotics/raspicam_node

tfoote, 2017-03-20:
it looks like you were ahead of the tutorials being filled out. The content in that file was only added 24 hours ago: https://github.com/UbiquityRobotics/rosdep/commit/05dc87c8666bd972aeca1b7f09cfbfb1afd07fd9

closed 2017-03-20 by tfoote