ros-infrastructure/rosdep #504: "ValueError: No JSON object could be decoded" when running `rosdep install --from-paths src --ignore-src --rosdistro indigo -y`

NataliaDiaz, 2017-04-04:
ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Bad installer [homebrew]: Error while parsing brew info for 'pyqt'
* Output of `brew info pyqt --json=v1`:

* Error while parsing:
Traceback (most recent call last):
File "/usr/local/lib/python2.7/site-packages/rosdep2/platforms/osx.py", line 184, in is_installed
pkg_info = json.loads(std_out)
File "/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 339, in loads
return _default_decoder.decode(s)
File "/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 364, in decode
obj, end = self.raw_decode(s, idx=_w(s, 0).end())
File "/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 382, in raw_decode
raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

wjwwood, 2017-04-04:
Can you try to update Homebrew and then see if the error persists?

```
$ brew update
```

NataliaDiaz, 2017-04-05:
The error persists...

wjwwood, 2017-04-05:
What command are you running and what version of Homebrew are you using?

NataliaDiaz, 2017-04-05:
brew update
Homebrew 1.1.12

wjwwood, 2017-04-05:
No, what command gives you the original traceback (I don't think `brew update` should give you that)?

NataliaDiaz, 2017-04-05:
The command is on the title of this issue. The output of brew info pyqt --json=v1   is
Error: No available formula with the name "pyqt" 

And if I do: 
`ros_catkin_ws$ brew install pyqt`
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (caskroom/cask).
No changes to formulae.

Error: No available formula with the name "pyqt" 
==> Searching for a previously deleted formula...
pyqt was deleted from homebrew/core in commit 2a6d326:
  pyqt: migrate to boneyard (https://github.com/Homebrew/homebrew-core/issues/6817)
  * pyqt: migrate to boneyard
  * gnuradio: remove pyqt, build docs by default.

To show the formula before removal run:
  git -C "$(brew --repo homebrew/core)" show 2a6d326^:Formula/pyqt.rb

If you still use this formula consider creating your own tap:
  http://docs.brew.sh/How-to-Create-and-Maintain-a-Tap.html

wjwwood, 2017-04-05:
I guess the issue is that Homebrew has removed `pyqt`. I don't know what the recommended strategy for installing `pyqt` should now be. I guess we'll have to add it to one of our taps.

wjwwood, 2017-04-05:
Once https://github.com/ros/homebrew-deps/pull/31 is merged, you should be able to do `brew update` one more time and have it work, assuming you have tapped ros/deps (it's in the installation instructions I think).

wjwwood, 2017-04-07:
@NataliaDiaz Since we're having trouble bring back the formula that Homebrew has recently removed, you might not be able to use Indigo anymore (since it needs qt version 4).

You can try using Kinetic in the meantime which uses qt version 5, which is still provided by Homebrew.

NataliaDiaz, 2017-04-09:
I am constrained to only use ROS Indigo and not Kinetic, because it is a requirement of Baxter robot arm simulator.
I re-executed brew install and now  I get: 
<details>
...
######################################################################## 100.0%
==> cmake . -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG 
==> make install
🍺  /usr/local/Cellar/urdfdom_headers/1.0.0: 23 files, 64.6KB, built in 26 seconds
==> Installing ros/deps/urdfdom 
==> Downloading https://github.com/ros/urdfdom/archive/1.0.0.tar.gz
==> Downloading from https://codeload.github.com/ros/urdfdom/tar.gz/1.0.0
######################################################################## 100.0%
==> cmake . -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG 
==> make install
🍺  /usr/local/Cellar/urdfdom/1.0.0: 19 files, 505.1KB, built in 55 seconds
executing command [sudo -H pip install -U netifaces]
Password:
Collecting netifaces
  Downloading netifaces-0.10.5.tar.gz
Building wheels for collected packages: netifaces
  Running setup.py bdist_wheel for netifaces ... done
  Stored in directory: /var/root/Library/Caches/pip/wheels/9f/40/bd/1f8e0f83e36399900d81bebfd7ece579931ced3a5d9383284b
Successfully built netifaces
Installing collected packages: netifaces
Successfully installed netifaces-0.10.5
executing command [brew install qt]
Updating Homebrew...
Warning: qt is a keg-only and another version is linked to opt.
Use `brew install --force` if you want to install this version
executing command [brew install gtest]
Updating Homebrew...
==> Installing gtest from ros/deps
==> Downloading https://github.com/google/googletest/archive/release-1.7.0.tar.g
==> Downloading from https://codeload.github.com/google/googletest/tar.gz/releas
######################################################################## 100.0%
==> Patching
patching file include/gtest/gtest.h
==> glibtoolize
==> aclocal
==> autoheader
==> automake --add-missing
==> autoconf
==> ./configure CPPFLAGS=-DGTEST_HAS_TR1_TUPLE=0 --prefix=/usr/local/Cellar/gtes
==> make
🍺  /usr/local/Cellar/gtest/1.7.0: 35 files, 1.2MB, built in 3 minutes 23 seconds
ERROR: the following rosdeps failed to install
  homebrew: Failed to detect successful installation of [qt]
</details>

Could `brew install --force qt` break it? is there another way to make work ROS Indigo in mac OS?
Thanks!

wjwwood, 2017-04-10:
> Could `brew install --force qt` break it?

Possibly? Probably, but I don't know. You're unfortunately off the beaten path now.

> is there another way to make work ROS Indigo in mac OS?

Unfortunately no. You can try using Kinetic anyways (the differences from Indigo may be minor enough to get you by) or you can try taking what we have so far in https://github.com/ros/homebrew-deps/pull/31 and trying to figure it out from there, or you can install a VM and use Ubuntu.

Sorry I couldn't be more help right now.

---

I'm going to close this since there's nothing to be done in `rosdep` other than improve the error message, but that's covered here already: https://github.com/ros-infrastructure/rosdep/issues/490

Also, from that issue, you can try skipping the rosdep keys that are giving you trouble, like `--skip-keys pyqt`, and just see if that gets enough stuff working for you. The reality is that only high level packages in ROS use qt, so you might be able to skip some things and get what you need working in the meantime.

I can reopen this later if needed.

closed 2017-04-10 by wjwwood

NataliaDiaz, 2017-04-14:
At least the message is more insightful now:
Even without the --with-qt5 option, you can display native VTK render windows
from python. Alternatively, you can integrate the RenderWindowInteractor
in PyQt4, Tk or Wx at runtime. Read more:
    import vtk.qt5; help(vtk.qt5) or import vtk.wx; help(vtk.wx)