ros-infrastructure/rosdep #507: rosdep update not working on Manjaro (arch - jade)

jaume-gasa, 2017-04-10:
```
rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"

ERROR: Rosdep experienced an error: Could not detect OS, tried ['windows', 'ubuntu', 'rhel', 'qnx', 'osx', 'opensuse', 'opensuse', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'debian', 'cygwin', 'centos', 'arch']
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.4

Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/site-packages/rosdep2/main.py", line 351, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/site-packages/rosdep2/main.py", line 359, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/site-packages/rosdep2/main.py", line 545, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/site-packages/rosdep2/sources_list.py", line 466, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/site-packages/rosdep2/gbpdistro_support.py", line 147, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python2.7/site-packages/rosdep2/__init__.py", line 85, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python2.7/site-packages/rosdep2/platforms/debian.py", line 50, in register_platforms
    register_linaro(context)
  File "/usr/lib/python2.7/site-packages/rosdep2/platforms/debian.py", line 63, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python2.7/site-packages/rosdep2/installers.py", line 112, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python2.7/site-packages/rospkg/os_detect.py", line 600, in get_name
    self.detect_os()
  File "/usr/lib/python2.7/site-packages/rospkg/os_detect.py", line 570, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s"%attempted)
OsNotDetected: Could not detect OS, tried ['windows', 'ubuntu', 'rhel', 'qnx', 'osx', 'opensuse', 'opensuse', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'debian', 'cygwin', 'centos', 'arch']
```

wjwwood, 2017-04-10:
You'd have to make a pull request against `rospkg` so that it can detect Manjaro as a rebranding of Arch.

This is similar to these `rospkg` pull requests:

- https://github.com/ros-infrastructure/rospkg/pull/109
- https://github.com/ros-infrastructure/rospkg/pull/82

@tfoote might also be able to give you pointers because he's dealt with OS aliasing in the past more than me.

wjwwood, 2017-04-10:
Btw, I closed this because action is required elsewhere. Feel free to continue to comment here and I can reopen if needed.

closed 2017-04-10 by wjwwood

tfoote, 2017-04-10:
Those are both good examples of how to add support.