ros-infrastructure/rosdep #508: How to use local rosdep location

davetcoleman, 2017-04-10:
I'm [following these instructions](https://github.com/ros-infrastructure/rosdep/blob/master/doc/contributing_rules.rst#point-your-sourceslistd-at-your-forked-repository) but I don't know how to make my rules point to a local filepath instead of a url...

I am trying to only change the base.yaml, below:
```
# os-specific listings first
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx

# generic
yaml file://home/dave/ros/standalone/rosdistro/rosdep/base.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
gbpdistro https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml fuerte

# newer distributions (Groovy, Hydro, ...) must not be listed anymore, they are being fetched fro$
```

But ``rosdep update`` says:

```
ERROR: unable to process source [file://home/dave/ros/standalone/rosdistro/rosdep/base.yaml]:
	<urlopen error [Errno 2] No such file or directory: '/dave/ros/standalone/rosdistro/rosdep/base.yaml'> (file://home/dave/ros/standalone/rosdistro/rosdep/base.yaml)
```

yet i can open that file externally

wjwwood, 2017-04-10:
I think you need `file:///abs/path` and not `file://abs/path`.

davetcoleman, 2017-04-11:
Documented https://github.com/ros-infrastructure/rosdep/pull/509/files

closed 2017-04-11 by davetcoleman