ros-infrastructure/rosdep #510: OSX homebrew "rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --as-root pip:yes --skip-keys "libqt5-core libqt5-gui libqt5-opengl libqt5-opengl-dev libqt5-widgets qt5-qmake qtbase5-dev" failed: ERROR: the following rosdeps failed to install   homebrew: Failed to detect successful installation of [pyqt5]

zhonghh, 2017-04-19:
OSX 10.12.4 Sierra
follow the instruction installing kinetic
because of the qt5 error I use:
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --as-root pip:yes --skip-keys "libqt5-core libqt5-gui libqt5-opengl libqt5-opengl-dev libqt5-widgets qt5-qmake qtbase5-dev"

However it stopped at:
Warning: pyqt-5.8.2 already installed
ERROR: the following rosdeps failed to install
  homebrew: Failed to detect successful installation of [pyqt5]

wjwwood, 2017-04-19:
If you're not installing qt5, then you won't have the right version of pyqt installed either.

Also see:

- https://github.com/ros-infrastructure/rosdep/issues/490
- https://github.com/ros-infrastructure/rosdep/issues/504
- https://github.com/ros/homebrew-deps/pull/31

zhonghh, 2017-04-20:
>    - ros/homebrew-deps#31 <https://github.com/ros/homebrew-deps/pull/31>

I have qt5 installed already. And I've already followed #490. This error
actually happens after that.

brew info qt returns:

qt: stable 5.8.0 (bottled), HEAD [keg-only]

Cross-platform application and UI framework

https://www.qt.io/

/usr/local/Cellar/qt/5.8.0_2 (8,667 files, 246.2MB)

  Poured from bottle on 2017-04-17 at 11:28:35

From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/qt.rb

==> *Dependencies*

Build: *pkg-config **✔*

==> *Requirements*

Build: *xcode **✔*

Required: *macOS >= 10.8 **✔*

Optional: *mysql **✘*, *postgresql **✘*

==> *Options*

--with-docs

Build documentation

--with-examples

Build examples

--with-mysql

Build with mysql support

--with-postgresql

Build with postgresql support

--with-qtwebkit

Build with QtWebkit module

--HEAD

Install HEAD version

==> *Caveats*

We agreed to the Qt opensource license for you.

If this is unacceptable you should uninstall.

This formula is keg-only, which means it was not symlinked into /usr/local.

Qt 5 has CMake issues when linked

If you need to have this software first in your PATH run:

  echo 'export PATH="/usr/local/opt/qt/bin:$PATH"' >> ~/.bash_profile

For compilers to find this software you may need to set:

    LDFLAGS:  -L/usr/local/opt/qt/lib

    CPPFLAGS: -I/usr/local/opt/qt/include

For pkg-config to find this software you may need to set:

    PKG_CONFIG_PATH: /usr/local/opt/qt/lib/pkgconfig

wjwwood, 2017-04-20:
Ok, well you'll have to skip pyqt as well. I guess the rosdep "installed detector" for Homebrew isn't handling the aliases correctly.

chexov, 2017-04-22:
I am new at rosdep but a little bit of digging in rosdep2/platforms/osx.py I found that ```brew list```
show qt5 dependency as just "qt" package. So basically qt5 is an alias for qt package according to brew. And it is never in ```installed_formulae``` list.

I disabled this quick check here https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/platforms/osx.py#L179 and it worked for me:)

@wjwwood I am not yet familiar with rosdep architecture to fix this correctly. ```brew list``` has ---versions parameter which can be used but its all magic... 

I would say that "qt5" is not a valid howebrew package and it should be changed to just "qt" in yamls since it's the same at this moment and everything will be working:)

chexov, 2017-04-22:
For all the travelers got in here. kinetic installing gazebo7 which requires qt4 which is a dead end. 
Found this for gazebo8 update https://github.com/ros/rosdistro/pull/14564

imvuong, 2017-04-28:
I got it to work by creating a symlink in /usr/local/Cellar
`cd /usr/local/Cellar`
`sudo ln -s pyqt pyqt5`

canlang, 2017-06-28:
@imvuong thanks! it work

wjwwood, 2017-07-28:
I think upstream changes in the rules for qt in Homebrew may have addressed this. Closing for now, feel free to continue commenting and I can reopen if necessary.

closed 2017-07-28 by wjwwood

maximkulkin, 2018-05-18:
Guys, I'll just leave it here:
```
brew install qt5
brew unlink qt@4
brew link --force qt
```