ros-infrastructure/rosdep #511: rosdep internal error: "brew info uuid"

danherman, 2017-04-28:
Following by rote the install steps for Kinetic on Mac OS X 10.11.6:
----------
finwe.local:/Projects/catkin_ws> rosdep install --from-paths src --ignore-src --rosdistro kinetic -y

ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Bad installer [homebrew]: Error while parsing brew info for 'uuid'
 * Output of `brew info uuid --json=v1`:
 
 * Error while parsing:
 Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/rosdep2/platforms/osx.py", line 184, in is_installed
    pkg_info = json.loads(std_out)
  File "/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
------

Note there is no such package:

finwe.local:/Projects/catkin_ws> brew info uuid
Error: No available formula with the name "uuid" 
==> Searching for a previously deleted formula...
Error: No previously deleted formula found.

There is however a homebrew package for ossp-uuid.
Yes, my homebrew is up to date and taps repaired.
No, I don't have a previous distro on tap.

finwe.local:~> brew tap
caskroom/cask
homebrew/boneyard
homebrew/core
homebrew/python
homebrew/science
homebrew/versions
osrf/simulation
ros/deps

danherman, 2017-04-28:
Leaving a solution to my own problem, in case it helps someone in the future.

I previously has ROS Indigo installed on this machine. Earlier during Kinetic installation I had ignored a warning:

bash-3.2$ sudo -H rosdep init
Password:
ERROR: default sources list file already exists:
	/etc/ros/rosdep/sources.list.d/20-default.list
Please delete if you wish to re-initialize

Beware ignoring warnings. Going back and deleting that file and proceeding everything works now:

sudo rm -rf /etc/ros/rosdep/sources.list.d/20-default.list
sudo -H rosdep init
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y

... and I'm back on track.

wjwwood, 2017-04-28:
Did you `rosdep update` before? That's more likely to have fixed the problem than removing and redoing `rosdep init`.

Please close this issue if you've worked around the problem.

danherman, 2017-04-28:
Actually I did rosdep update, so that by itself wasn't enough. Thanks, closing.

closed 2017-04-28 by danherman