ros-infrastructure/rosdep #512: "Could not detect OS" when running "rosdep update" on Deepin OS 15.4 (Debian)

jingedawang, 2017-05-09:
Deepin OS is a distribution of Debian. See https://www.deepin.org/en/

when running `rosdep update`, I got

```
wjg@wjg-PC:~/Desktop$ rosdep update

reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"

ERROR: Rosdep experienced an error: Could not detect OS, tried ['windows', 'ubuntu', 'slackware', 'rhel', 'qnx', 'osx', 'opensuse', 'opensuse', 'neon', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'debian', 'cygwin', 'centos', 'arch']
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.4

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 351, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 359, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 545, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 466, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 147, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python2.7/dist-packages/rosdep2/__init__.py", line 85, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.py", line 50, in register_platforms
    register_linaro(context)
  File "/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.py", line 63, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 112, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python2.7/dist-packages/rospkg/os_detect.py", line 622, in get_name
    self.detect_os()
  File "/usr/lib/python2.7/dist-packages/rospkg/os_detect.py", line 592, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s"%attempted)
OsNotDetected: Could not detect OS, tried ['windows', 'ubuntu', 'slackware', 'rhel', 'qnx', 'osx', 'opensuse', 'opensuse', 'neon', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'debian', 'cygwin', 'centos', 'arch']
```

tfoote, 2017-05-09:
Deepin is not a debian variant we currently have support for. Please see the developer guide for how to contribute os detection for Deepin: http://docs.ros.org/independent/api/rosdep/html/developers_guide.html

An example of adding an OS detector is here: https://github.com/ros-infrastructure/rospkg/pull/85

And an example of adding an aliased release: https://github.com/ros-infrastructure/rosdep/pull/370

wjwwood, 2017-07-28:
Closing because it seems @tfoote directed @jingedawang to the way to move forward (though not an immediate solution). I can reopen when/if necessary.

closed 2017-07-28 by wjwwood

trDjango, 2017-11-26:
it will be really good if a deepin linux operating system is added. When will it be added deeply to support the operating system?when to support this operating system, a road map is available?(deepin linux)

tfoote, 2017-11-26:
There is no target or roadmap to support deepin. As linked above we will accept contributions to add support for OSs from the community. If you'd like to see support for it please look through the links above and consider opening a pull request to add support.

BinAddress, 2019-03-18:
希望加入对deepin系统的支持，或推出相应的补丁包，非常喜欢deepin的桌面环境！！！

tfoote, 2019-03-18:
@BinAddress A contribution of support for Deepin would be appreciated. The above thread mentions what is necessary to add support.