ros-infrastructure/rosdep #513: OSX Unable to resolve dependencies [pyqt]

budekatude, 2017-05-13:
I've been following the install instructions for Indigo on Mac OSX in the ROS wiki. Everything works fine until I try to resolve the dependencies:

`Matty:ros_catkin_ws Matthias$ rosdep install --from-paths src --ignore-src --rosdistro indigo -y`

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
python_qt_binding: No definition of [python-qt-bindings] for OS [osx]
qwt_dependency: No definition of [python-qt-bindings-qwt5] for OS [osx]
rviz: No definition of [libqt4-opengl-dev] for OS [osx]
gl_dependency: No definition of [python-qt-bindings-gl] for OS [osx]
webkit_dependency: No definition of [python-qt-bindings-webkit] for OS [osx]
qt_gui_cpp: No definition of [qt4-qmake] for OS [osx]
qt_gui: No definition of [python-qt-bindings] for OS [osx]
turtlesim: No definition of [libqt4] for OS [osx]
Matty:ros_catkin_ws Matthias$ cd /usr/local/Cellar
Matty:Cellar Matthias$ sudo ln -s pyqt5 pyqt
```

I tried manually installing pyqt from homebrew and this is the status I get:

```
pyqt: stable 5.8.2 (bottled)
Python bindings for v5 of Qt
https://www.riverbankcomputing.com/software/pyqt/download5
/usr/local/Cellar/pyqt/5.8.2 (856 files, 24.7MB) *
  Poured from bottle on 2017-05-13 at 15:08:46
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/pyqt.rb
==> Dependencies
Required: qt ✔, sip --with-python3 ✔
==> Requirements
Recommended: python3 ✔
Optional: python ✔
==> Options
--with-debug
	Build with debug symbols
--with-docs
	Install HTML documentation and python examples
--with-python
	Build with python support
--without-python3
	Build without python3 support
```
I've also tried relinking pyqt, without resolving the problem.

Any help is highly appreciated, sorry if I missed something... fairly new to this.
Thanks.

wjwwood, 2017-05-15:
Indigo is not going to work any longer because Homebrew removed all of the `qt4` related formulae:

https://github.com/ros/homebrew-deps/pull/31

wjwwood, 2017-05-15:
You can try Kinetic, it will use qt5 and should just work unless something else changed upstream.

closed 2017-05-15 by wjwwood

QinZiwen, 2018-07-20:
@budekatude @wjwwood . I try kinetic to Mac, But the page show that "This is a work in progress! It really won't work right now...".
When I rosdep install indigo, some errors show:
```
$ rosdep install --from-paths src --ignore-src --rosdistro indigo -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
python_qt_binding: No definition of [python-qt-bindings] for OS [osx]
qwt_dependency: No definition of [python-qt-bindings-qwt5] for OS [osx]
rviz: No definition of [libqt4-opengl-dev] for OS [osx]
gl_dependency: No definition of [python-qt-bindings-gl] for OS [osx]
webkit_dependency: No definition of [python-qt-bindings-webkit] for OS [osx]
qt_gui_cpp: No definition of [qt4-qmake] for OS [osx]
qt_gui: No definition of [python-qt-bindings] for OS [osx]
turtlesim: No definition of [libqt4] for OS [osx]
actionlib: No definition of [python-wxtools] for OS [osx]
```

hanqingsun, 2019-06-20:
anybody solve the problem?