ros-infrastructure/rosdep #519: Can't choose whether or not to install pip2 or pip3 version of pip package

jlack1987, 2017-06-13:
Have run into the case where I need a specific version of a pip, but it looks like the in the rosdep yaml files rosdep is not setup to look explicitly for either `pip2` or `pip3` and instead just looks for `pip` and uses whatever version the `pip` executable is using. It would be awesome if we could explicitly say a key should be installed with `pip2 ` or `pip3`

wjwwood, 2017-07-17:
Sounds good to me. I don't have time to look into it myself, but I'll try to review and release a pr that adds a `pip3` installer.

willcbaker, 2017-08-08:
agreed; can't use rosdep if you have pip3 installed. lame

willcbaker, 2017-08-18:
@wjwwood any comment on how best to implement this? I added a test-passing pip2/pip3 installer but am unaware of how to run the tests locally

wjwwood, 2017-08-21:
I would drop the `pip` generic installer and invoke pip via python, i.e. `python2 -m pip ...` or `python3 -m pip ...`, where appropriate.

mikepurvis, 2017-10-19:
I agree with that direction, but note that it gets hairy on OS X, as homebrew now installs its brewed pythons as `python2` and `python3`, leaving the system python as `python`.

I'm not sure what the best way out of this is, except that there need to be multiple ways to override which python is used to invoke `pip`.

PymZoR, 2019-10-02:
As I read http://wiki.ros.org/UsingPython3/IdentifyDependencies, in particular the example on how to add Python3 packages rosdep key (https://github.com/ros/rosdistro/pull/21800), I wondered how to add `pip3` dependencies. 

As Noetic is coming and support for Python3 packages can be quite limited if only `pip` and system package managers are available on rosdep, is the `pip3` installer support considered ?

tfoote, 2019-10-16:
With the new ROS_PYTHON_VERSION definitions in [REP 149](https://www.ros.org/reps/rep-0149.html) I think that it makes sense that all the pip rules should resolve to pip3 for noetic by default unless the user sets ROS_PYTHON_VERSION manually. In which case the pip installer will use the user's chosen version.  #709 is going in that direction.

PR #709 (2019-10-18): Pip respects ROS_PYTHON_VERSION
~Requires #708~ **Merged**
Resolves #519

Unlike other rosdep keys, pip-only rosdep keys are not separated by python version. Currently rosdep calls the command `pip`, which seems to commonly be Python 2, but may be Python 3.

As recommended in ros-infrastructure/rep#201, this makes the pip installer respect `ROS_PYTHON_VERSION`. I added more fall-back options than originally listed in that proposal.

First this PR tries to use `pip2` or `pip3` if those exist. I put this as the first option since it's closest to the current use of `pip`. Second, if the requested [... cut]
closed 2019-10-18 by sloretz