ros-infrastructure/rosdep #522: rosdep init error InvalidData: line:

rlk7624, 2017-07-12:
rosdep version: 0.11.5

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 351, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 359, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 486, in command_init
    data = download_default_sources_list()
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 325, in download_default_sources_list
    parse_sources_data(data)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 366, in parse_sources_data
    raise InvalidData("line:\n\t%s\n%s"%(line, e), origin=origin)
InvalidData: line:
	<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01//EN" "http://www.w3.org/TR/html4/strict.dtd">
type must be one of [yaml,gbpdistro]

tfoote, 2017-07-12:
This looks like you have invalid data in one of your rosdep sources. It looks like the contents might be empty.

rlk7624, 2017-07-12:
how to solve this issue?

tfoote, 2017-07-12:
Make sure that your rosdep sources list is valid. http://docs.ros.org/independent/api/rosdep/html/sources_list.html

You should have something like this: 
```
$ cat /etc/ros/rosdep/sources.list.d/20-default.list 
# os-specific listings first
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx

# generic
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
gbpdistro https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml fuerte

# newer distributions (Groovy, Hydro, ...) must not be listed anymore, they are being fetched from the rosdistro index.yaml instead

```

rlk7624, 2017-07-12:
list is valid

tfoote, 2017-07-12:
do you have any other lists in the same directory?

rlk7624, 2017-07-12:
NO

tfoote, 2017-07-12:
Looking closer at your traceback I believe that you're calling `rosdep init` which is trying to dowload the data from `DEFAULT_SOURCES_LIST_URL = 'https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list'` and then the resulting data is failing to be parsed.

This error is usually caused by a firewall or network configuration that prevents pulling the content from that url. If you need more help please ask for help on http://answers.ros.org and include a full description of how you have setup your computer, what your network is like. And what commands you're running with the full console output not just the traceback.

closed 2017-07-12 by tfoote