ros-infrastructure/rosdep #524: ERROR: Rosdep experienced an error: sequence item 0: expected string, HomebrewResolution found

pitosalas, 2017-07-18:
ERROR: Rosdep experienced an error: sequence item 0: expected string, HomebrewResolution found
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 353, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 447, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 634, in command_install
    print("uninstalled dependencies are: [%s]"%(', '.join([', '.join(pkg) for pkg in [v for k,v in uninstalled]])))
TypeError: sequence item 0: expected string, HomebrewResolution found

bash-3.2$ rosdep install -v --from-paths src --ignore-src --rosdistro kinetic -y --skip-keys "libqt5-core libqt5-gui libqt5-opengl libqt5-opengl-dev libqt5-widgets qt5-qmake qtbase5-dev python-qt5-bindings-webkit"

COMPLETE CONSOLE: https://pastebin.com/51LHG1UY

wjwwood, 2017-07-18:
I was able to reproduce this using a similar command to yours. I made https://github.com/ros-infrastructure/rosdep/pull/525 to fix the issue (it's just a bug in the printing for verbose mode).

If you're able to try out the pr that would be great feedback.

closed 2017-07-18 by wjwwood

pitosalas, 2017-07-19:
How do I try it out? How do I pull in the latest into my environment?

Pito Salas
Brandeis Computer Science
Feldberg 131

wjwwood, 2017-07-24:
Clone this repository and run `cd path/to/cloned/repository && python setup.py install`. You can uninstall the custom version with `python -m pip uninstall rosdep`.