ros-infrastructure/rosdep #526: Support for replace and conflict tag

Baycken, 2017-07-19:
Hello,
As per [REP 127](http://www.ros.org/reps/rep-0127.html#id69), package.xml now has `replace` and `conflict` tag. I am wondering when those tags will be supported by rosdep.

NikolausDemmel, 2017-07-19:
What use case do you have in mind? What would you want rosdep to do?

Baycken, 2017-07-19:
Basically rosdep would skip a key if the key is replaced by another package. If a key is conflicting with a another package, rosdep would also skip the key and warn the user about it.

Baycken, 2017-07-19:
I have an idea of how to do this and can help code it if needed.

NikolausDemmel, 2017-07-19:
I don't think there are plans yet to support this, but I'm sure pull requests are welcome if you want to take a stab at implementing this feature.

> Basically rosdep would skip a key if the key is replaced by another package. If a key is conflicting with a another package, rosdep would also skip the key and warn the user about it.

Not that there are many cases to consider here. Just some off the bat:

- what if a rosdep key marked as conflicting is already installed?
- what if a rosdep key makred as conflicting is satisfied by a from-source ros package in your workspace?
- what if there are multiple candidates to be installed that are mutually conflicting? Or if package A requires system dependecy with rosdep A-dep and package B requires ROS package B-dep, which is therefore to be installed (via binary distribution), but now B-dep says it conflicts with A-dep?
- similar situations with `replace` might occur.

I don't want to discourage, just saying that the semantic in various situations should be carefully considered.

Baycken, 2017-07-19:
@NikolausDemmel  can you have a look at my PR?

NikolausDemmel, 2017-07-20:
> This could also serve as a more permanent alternative to --skip-keys by allowing packages to blacklist keys of other packages.

Btw, just to add to this, there already is a permanent way to always skip certain keys. See the documentation of `--skip-keys` (http://docs.ros.org/independent/api/rosdep/html/commands.html).

Baycken, 2017-07-20:
Yeah, I knew that, but that method is machine specific fix and not portable.

wjwwood, 2017-07-24:
> Yeah, I knew that, but that method is machine specific fix and not portable.

How so?

Baycken, 2017-07-25:
So you will need to change the base.yaml file to change the definition of that specific package right? So this fix only affect the machine which the `sources.list.d` is modified. While this process can be automated, it's effect still affect that machine only. For organization with many developers or robot deployment, it might be not so good of a solution.

tfoote, 2017-07-25:
@Baycken Yes, it needs to be added to your robot deployment mechanism and your developer environment setup instructions. It's adding one command to the usual 10 or so installation steps. This level of configuration is standard for robot distributors who add custom sources etc.