ros-infrastructure/rosdep #528: rosdep creates files under root wheel despite sudo -H

pitosalas, 2017-07-21:
This doesn't make sense to me, is there a way this could be a bug in rosdep?

```
~ ls /etc/ros/rosdep/sources.list.d
10-ros-install-osx.list
~  sudo -H rosdep init
Wrote /etc/ros/rosdep/sources.list.d/20-default.list
Recommended: please run

	rosdep update

~ ls -l /etc/ros/rosdep/sources.list.d
total 16
-rw-r--r--  1 root  wheel   90 Jul 18 15:56 10-ros-install-osx.list
-rw-r--r--  1 root  wheel  588 Jul 20 20:47 20-default.list
```

Note that 20-default.list is created as root wheel. Is that ok?

wjwwood, 2017-07-21:
`sudo -H` just sets your home directory I think. It doesn't affect the owner of the files created or anything like that. The default location for the settings (i.e. `/etc/ros/rosdep/sources.list.d`) is not affected by the home directory.

I don't really see this as a bug, but I guess I could be wrong.

pitosalas, 2017-07-21:
I could be mistaken. It just caught my eye and thought I would post it. Here's the doc:

`sudo -H: Request that the security policy set the HOME environment variable to the home directory specified by the target user's password database entry. Depending on the policy, this may be the default behavior.
`
Which I thought meant that files and directories created under that would belong to current user. That might not be a correct belief.

Never mind, I think.

wjwwood, 2017-07-21:
Yeah, I think it _just_ sets the `HOME` environment variable. I'll close this for now. Feel free to continue the discussion though.

closed 2017-07-21 by wjwwood

NikolausDemmel, 2017-07-21:
I agree with @wjwwood.

Also note that if anything, it would be the other way around. The target user for sudo in this case is `root`. At least on Ubuntu the default behaviour of `sudo` (without `-H`) is not changing the the home directory, i.e. you execute the command with the home directory set to the current user, and using `sudo -H` means changing `HOME` to `/root`.