ros-infrastructure/rosdep #532: v1.1.6 breaks support for hydro

ipa-mdl, 2017-07-28:
The new version does not work on hydro anymore, because  `apt show` is used for detecting virtual packages, but `apt` ist not present.

How did v1.1.6 end up in the hydro release?

wjwwood, 2017-07-28:
The current version is `0.11.6`. 
`rosdep` isn't tied to a single ros distro, the same version is used globally. 

I'll fix this issue if I can, I need to look at your pr which I think is related, but hydro is long sense eol and so are the ubuntu versions that hydro was supported on.

ipa-mdl, 2017-07-28:
>  rosdep isn't tied to a single ros distro, the same version is used globally.
>  but hydro is long sense eol and so are the ubuntu versions that hydro was supported on.

Why do new version get released to EOL distros?
The last (working) version should be kept.

> I'll fix this issue if I can, I need to look at your pr which I think is related

It should at least fix the `apt show` issue.

tfoote, 2017-07-28:
We can turn off releases for older distros. However there's two issues. One is that sequential rosdistros have overlapping Ubuntu distros so it's not really a binary as to which one gets released. And the other major thing is that things like rosdep interact with online resources we maintain. So if there's ever a change in the hosting the packages may need to be updated, thus the last "working" version will potentially stop working.

We also know that there are a lot of people using the EOL'd distros still. And some of these features will be valuable to them. 

With respect to this specific change `apt` specifically recommends against using it in automation, giving a warning that it's interface is not considered stable. Thus switching to apt-cache is a better solution even for the newer platforms.

wjwwood, 2017-07-31:
I no longer have a precise machine to test with, but https://github.com/ros-infrastructure/rosdep/pull/533 works with trusty and xenial, so I'll close this for now and do a new release this afternoon.

closed 2017-07-31 by wjwwood

ipa-mdl, 2017-07-31:
I will test with `industrial_ci` again after it was released! :)