ros-infrastructure/rosdep #534: v0.11.7 breaks rosdep for meta-packages

ipa-mdl, 2017-08-02:
Looks like #531 was not complete.

If rosdep fails with 
```
MY_METAPACKAGE: Missing resource MY_METAPACKAGE
```

I will look into it again.

ipa-mdl, 2017-08-03:
fixed in #535

closed 2017-08-04 by wjwwood

wjwwood, 2017-08-04:
`0.11.8` is out

ipa-mdl, 2017-08-04:
There is still an issue:
Some of our meta-packages depend on `python-wstool`.
For distros < kinetic it is listed in https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml#L3629 and it works as a fall-back.
For newer distro the key for `python-wstool` gets read from somewhere else, but where?

wjwwood, 2017-08-07:
> I guess the key needs to be added. We did not discover this yet because meta-packages got skipped.

I guess so, it should be in all the OS's as `python-wstool`. I'll let you open a pr against ros/rosdistro if it's needed.

P.S. prefer a new comment to editing when adding info (we don't get notifications of an edit)

PR #535 (2017-08-04): fix get_view_key for meta-packages
fixes #534 

Should the stack keys get cached as well?