ros-infrastructure/rosdep #536: rosdep 0.11.6/7 does not work on precise

k-okada, 2017-08-02:
latest update to rosdep 0.11.6 changes the precise binary(http://repositories.ros.org/ubuntu/main/dists/precise/main/binary-amd64/Packages) and 
we have errors, something like
```
root@f93d6ca4d058:~# rosdep install --from-paths ./ros_tutorials/ --rosdistro hydro
No packages found matching qt5-qmake.
No packages found matching ros-hydro-message-generation.
No packages found matching ros-hydro-roscpp-serialization.
No packages found matching ros-hydro-rosconsole.
No packages found matching ros-hydro-rospy.
No packages found matching ros-hydro-message-runtime.
No packages found matching qtbase5-dev.
No packages found matching ros-hydro-roscpp.
No packages found matching ros-hydro-geometry-msgs.
No packages found matching ros-hydro-roslib.
```
If we roll back the rosdep to 0.11.5 with pip command, the issue was disappeared.

ipa-mdl, 2017-08-02:
The hydro problem was already reported in #532 and fixed in #533.
This fix was released as v.0.11.7, but introduced a regression #534.

k-okada, 2017-08-03:
thank you for explanation, waiting for 0.11.8 includes #534

closed 2017-08-03 by k-okada