ros-infrastructure/rosdep #539: Ros depedencies does not fallback to ROS packages

yotabits, 2017-08-08:
@tfoote @dirk-thomas @AlexV @yichoe
I have a new issue with the python-watchdog package, it's not found any more on trusty.

`rosdep resolve python-watchdog
`

Has the following output

 `ERROR:`
 `No definition of [python-watchdog] for OS version [trusty]`
 `No definition of [python-watchdog] for OS version [trusty]`
        `rosdep key : python-watchdog`
        `OS name    : ubuntu`
        `OS version : trusty`
        `Data: _is_ros: true`
`debian:`
`- python-watchdog`
`fedora:`
  `21:`
    `yum:`
      `packages:`
      `- ros-indigo-python-watchdog`
  `heisenbug:`
    `yum:`
      `packages:`
      `- ros-indigo-python-watchdog`
`osx:`
  `homebrew:`
    `packages:`
    `- ros/indigo/python-watchdog`
`ubuntu:`
  `artful:`
  '- python-watchdog`
  xenial:`
  `- python-watchdog`
  `yakkety:`
  `- python-watchdog`
  `zesty:`
  `- python-watchdog`

It seems like when the system package is not found it does not fall back to the ROS package.

tfoote, 2017-08-08:
I was unable to reproduce this until I ran rosdep update after which I get the same error: 

https://gist.github.com/tfoote/ba6d8a28265b9052f2b7f6c2867f2864

python-watchdog rules were recently added for newer platforms: https://github.com/ros/rosdistro/pull/15578

yotabits, 2017-08-08:
@tfoote Thank you for taking the time to test :+1:  
I don't know exactly how rosdep works but after some debugging I found out that:

`rosdep resolve {pakage-name}`

result in a call to the [command_resolve function](https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/main.py#L783)
This function "load" the following yaml from .ros/rosdep/source.cache:
                       
`  yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml`                                     
`  yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml`                                                       
`  yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml`

python-watchdog is then found under:
`  yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml` 

But as long as  my actual Os (Trusty) is not in the python.yaml file for the python-watchdog
this [function call](https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/main.py#L802) raise an error.

A quick walk around would be to add:
`trusty: [ros-indigo-python-watchdog]`
But it's kind of break the logic

NikolausDemmel, 2017-08-08:
The current implementation stops looking in other sources, when a rosdep key is found in one, regardless if it has a resolution for the current os or not. AFAIK there is no rule merging. Or is the rule merging applied with a given OS/VERSION, such that the rule entry is actually empty on some platforms (e.g. trusty in this case)?

With the former, rosdep rules like ros/rosdistro#15578 break older platforms.

Or is it the latter, and there is a bug that makes this fail still?

asmodehn, 2017-08-10:
@NikolausDemmel I would says that having PR like https://github.com/ros/rosdistro/pull/15578 break older platforms is not a suitable behavior.

Otherwise, we potentially would have to duplicate all ros packages as rosdep keys, for each package in each distro, to prevent accidental future breakage...

NikolausDemmel, 2017-08-10:
I'm not saying it should not be changed. However, it might amount to a substantial change in semantics of rosdep rules.

Former ros packages that become system dependencies in later distributions are tricky. It is maybe not even clear what rosdep should do. Consider a Ros package foo in indigo that is now a system dependency foo in kinetic. Now you add rosdep rules like above for xenial and up. But when someone installs indigo on xenial (from source) and runs rosdep on something depending on foo, then rosdep would install foo from apt, when it should rather try to install "ros-indigo-foo" (which apt would fail to find, since there are no indigo binaries on xenial, meaning you should get it from source).

To fix this you'd need some notion of rosdistro in rosdep rules, which AFAIK there is not.

asmodehn, 2017-08-11:
Interesting example... but Indigo Igloo is not supported on xenial, so we do not need to guarantee anything there. And actually **I would have expected rosdep to install foo from apt in that case**. If someone is trying to port some indigo/trusty software to xenial, he/she should get the source of everything that needs to be ported, dependencies included, or rely on the packages from the target platform.

I personally use rosdep as a "cross-platform package manager", not as a "rosdistro integrity checker". I think it is actually good that rosdep doesn't have any notion of a ROS distribution, the distribution should be irrelevant from the package manager point of view...

Rosdep is supposed to work on any platform, and the rules are here to provide mapping between a name and a package on a platform. So I think this should be addressed without changing much in the semantics of the rules. I would rather see the expectation of rosdep managing ros packages reduced to the minimum...

I think (_disclaimer : I haven't had a look at the code_) the behavior of matching a rule name with a ROS package, should be a small, on the side, error handling feature of rosdep: "If I cannot find any package name specified in my rules, maybe it s a ROS package? Lets try to append `ros-<distro>-` and get it from local system package manager. ". But apart from that, rosdep should be as generic as possible, because it matches what I experience all the time as a user : 

- rosdep install my dependencies
- build&run -> fail
- add a rule in rosdistro/rosdep
- rosdep install my dependencies
- build&run -> success

Compared to that flow, this one is much more rare (and doesn't involve any rule as far as I can see): 
- rosdep install my dependencies
- build&run -> fail -> "Oh I forgot to include a ROS package in my source and it's not released in my rosdistro"
- ~~release~~ too long -> include package in source
- build&run -> success (no rosdep required)

yotabits, 2017-08-17:
@tfoote Any clue on the strategy to adopt ?

tfoote, 2017-08-17:
We historically have tried to keep rosdep keys and packages keys unique and non-overlapping.

Recently we've worked to add support for having keys so that packages can transition from system packages into ROS packages. I think @NikolausDemmel might be right that there's an issue with the rule merging at too high a level. If it continued searching down to the level of the OS/Version this would be able to resolve to the ROS package since the rosdep is not defined on Trusty.

The real challenge here is that if we have potentially two packages with the same symbols etc. One on the system and one in the ROS tree there's a potential for collisions, in this case it's python so it's not too problematic. But we really don't want to have ROS packages masking system libraries at different versions.

To avoid the above conflict I think we need to keep the rosdep rule as taking priority over a ROS package. So that the ROS package is effectively only used when there is not a conflicting rosdep key. And it's likely that we should add logic to our other release and build pipelines to have this same masking behavior. That gets way more complicated, and has ramifications if there are changed rosdep rules.

wjwwood, 2017-10-05:
So, what's the conclusion here, should we say that we won't change the behavior (after briefly reading I think that's the best solution) and if so what's the work around?

tfoote, 2017-10-05:
I think that the current behavior of the rosdep overriding the ros package in the resolution order. We should work hard to never have a platform where there is both a ROS package and a system package with the same name.

What needs to change to resolve this issue is that when rosdep finds a top level key, but cannot resolve it for the specific platform it does not continue to search for the resolution in the later sources (aka the rosdistro source).

As a workaround for anyone experiencing this they can add a local rosdep rules file on their computer which will resolve to the rosdistro version of the apt package.

NikolausDemmel, 2017-10-06:
> Interesting example... but Indigo Igloo is not supported on xenial, so we do not need to guarantee anything there. And actually I would have expected rosdep to install foo from apt in that case. If [...]

Fair enough, the example was not great admittedly.

> I personally use rosdep as a "cross-platform package manager", not as a "rosdistro integrity checker". I think it is actually good that rosdep doesn't have any notion of a ROS distribution, the distribution should be irrelevant from the package manager point of view...

I agree. My point was, you cannot in general put rules like

```
 trusty: [ros-indigo-python-watchdog]
```

as workarounds in rosdep files, since there is frequently more than one ROS distro supported on some versions of Ubuntu. I'm not saying it is a good idea, but that is why you'd need a notion of rosdistros in rosdep files to resort to such workarounds (except if you only do it locally). But we all agree we don't want that.

> Rosdep is supposed to work on any platform, and the rules are here to provide mapping between a name and a package on a platform. So I think this should be addressed without changing much in the [...]

Agreed, but addressing this requires changing the merging behaviour, as Tully summed up, and that is a change in semantics.

> I think (disclaimer : I haven't had a look at the code) the behavior of matching a rule name with a ROS package, should be a small, on the side, error handling feature of rosdep: "If I cannot find [...]

AFAIK this is exactly what is happening in the current implementation. The only problem is that currently finding any rule in one source stops the search in the sources "below". At this point in the code the (OS,version) tuple is not considered at all, yet.

Basically, IMHO @tfoote summed up perfectly in the previous comment what should be done. We just have to make sure that this change does not have any unintended side effects in other scenarios.