ros-infrastructure/rosdep #546: Unable to run tests following steps in documentation

willcbaker, 2017-08-18:
According to http://docs.ros.org/independent/api/rosdep/html/developers_guide.html

to run a test 
```cd rosdep2/test
nosetests```

but there is no test folder under rosdep2.
If I run nosetests in the top test folder all tests fail; documentation to run and add new tests is outdated

tfoote, 2017-08-18:
You run the nosetests as the root of the repo. That documentation appears to be out of date. A PR to fix that would be appreciated.

wjwwood, 2017-10-05:
There is a folder there, the code you're referencing assumes you clone to `rosdep2` and the `test` folder is in the root of the repository:

https://github.com/ros-infrastructure/rosdep/tree/master/test

It probably works either from the root of the repository (which is what our travis does, see: https://github.com/ros-infrastructure/rosdep/blob/29eb3633bfe2857afc3919405fa66fffb669d2ba/.travis.yml#L14) or from the test directory as the documentation says. Feel free to make a pr to update or clarify the docs, but I'll close this for now.

closed 2017-10-05 by wjwwood