ros-infrastructure/rosdep #548: Rosdep Errno 13 - on Snapdragon flight

joeycahill, 2017-08-30:
Hello, I was following the process found [here](https://github.com/ATLFlight/ATLFlightDocs/blob/master/SnapdragonROSInstallation.md) to install ROS on the Snapdragon flight, when I came to the rosdep update command I mistakenly ran it while root. Now I get this error:

ERROR: Rosdep experienced an error: [Errno 13] Permission denied: '/home/linaro/.ros'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.8

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 137, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 381, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 389, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 579, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 459, in update_sources_list
    retval.append((source, write_cache_file(sources_cache_dir, source.url, rosdep_data)))
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 537, in write_cache_file
    os.makedirs(source_cache_d)
  File "/usr/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/usr/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/home/linaro/.ros'

I would appreceiate any help.

Thank you

wjwwood, 2017-08-30:
> `OSError: [Errno 13] Permission denied: '/home/linaro/.ros'`

Sounds like a permission issue on that folder. Make sure you have permission to read/write to that folder.

closed 2017-08-30 by wjwwood

joeycahill, 2017-08-30:
When I started the tutorial I was root of the system. When I ran rosdep update it warned that root was not the best way to run rosdep update. So I canceled the process and exited root and tried again. That is when the issue happened. To fix it I became root again and ran rosdep update. Everything went ok as far as I can tell.

tfoote, 2017-08-30:
Instead of running as root again. You should `chown` the folder to be owned by your user, i'm guessing `linaro` so you don't need to run as root in the future.

joeycahill, 2017-08-31:
Thank you very much, I will try that out.