ros-infrastructure/rosdep #549: When I compile “ORB_SLAM2/Examples/ROS”, I encountered this problem

fighterxi, 2017-08-31:
sun@sun-Aspire-4743:~/orbslamv2/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build$ cmake .. -DROS_BUILD_TYPE=Release  
[rosbuild] Building package ORB_SLAM2
[rosbuild] Cached build flags older than manifests; calling rospack to get flags
Failed to invoke /opt/ros/indigo/bin/rospack cflags-only-I;--deps-only ORB_SLAM2
Package roscpp was not found in the pkg-config search path.
Perhaps you should add the directory containing `roscpp.pc'
to the PKG_CONFIG_PATH environment variable
No package 'roscpp' found
[rospack] Error: python function 'rosdep2.rospack.call_pkg_config' could not call 'pkg-config --cflags-only-I roscpp' without errors

CMake Error at /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:129 (message):
  

  Failed to invoke rospack to get compile flags for package 'ORB_SLAM2'.
  Look above for errors from rospack itself.  Aborting.  Please fix the
  broken dependency!

Call Stack (most recent call first):
  /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:231 (rosbuild_invoke_rospack)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/sun/orbslamv2/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".

____________________________________________________________
thsis is in  my   bashrc,and i think it's no problem with my ROS_PACKAGE_PATH.
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/sun/orbslamv2/src/ORB_SLAM2/Examples/ROS 
Has anyone ever encountered this problem?

fighterxi, 2017-08-31:
My LINUX version ：ubuntu 14.04  indigo

fighterxi, 2017-09-09:
I changed to another ubuntu 14.04 and i never found such probelm.

tfoote, 2017-09-09:
Thanks for letting us know. Since it's not reproducible I'm going to close this.

closed 2017-09-09 by tfoote