ros-infrastructure/rosdep #550: rosdep installation failed with pygraphviz install

Edith001, 2017-09-17:
Hi I am running rosdep installation like this "rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --skip-keys "libqt5-core libqt5-gui libqt5-opengl libqt5-opengl-dev libqt5-widgets qt5-qmake qtbase5-dev" and returned with  failed while pip install pygraphviz failed.The terminal returns as shown in the picture.
[image]
Could anyone please help me with that?

wjwwood, 2017-09-18:
Please do not post a picture of text. Update your issue with the text in the error (including more that what is in the picture) and I will try to help.

In general, never, ever, ever post a picture of text. It cannot be searched or copied and pasted.

----

Do you have graphviz installed from homebrew?

wjwwood, 2017-09-18:
Closing for now, I will reopen this when the original issue is updated.

closed 2017-09-18 by wjwwood