ros-infrastructure/rosdep #554: debian.py check for virtual packages is slow

jacobperron, 2017-10-04:
In *debian.py* checking for remaining virtual packages one-by-one in a loop at [L189](https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/platforms/debian.py#L189) is quite slow for large lists of packages. This is due to the `apt-cache show` command. In particular, I'm looking to cut down on build times in some of my pipelines.

I've noticed that calling `apt-cache show` on packages sequentially is a lot slower than passing all packages at once. Example execution times for 58 packages from the shell:

```
time cat test_pkgs | xargs -L1 apt-cache show
```
**1m 12.915s**

```
time apt-cache show $(cat test_pkgs)
```
**0m 2.920s**

I'm exploring possibilities for enhancement of the Python script.

As an aside, I noticed that packages are checked twice over. Does this have something to do with separating build and run depends?

Nickolaim, 2018-03-18:
@jacobperron  - is the issue still valid?  Pull request #521 touches on this functionality, and I see in the code `cmd = ['apt-cache', 'showpkg'] + list(packages)` that is probably doing what was requested in the issue.

If it is not the case, please provide a repro, I will be happy to work on it.

jacobperron, 2018-03-23:
@Nickolaim Does not appear to be an issue any more. Cheers!

closed 2018-03-23 by jacobperron