ros-infrastructure/rosdep #558: 'NoneType' object has no attribute 'strip'

sumitjagdale, 2017-10-12:
I am trying to install moveit libraries from source on ros version kinetic. 
$rosdep install -y --from-paths . --ignore-src --rosdistro kinetic
 
Experienced an error as follow: 
ERROR: Rosdep experienced an error: 'NoneType' object has no attribute 'strip'

Thanks in advance!

wjwwood, 2017-10-12:
I need the traceback if there is one. That is not enough information to easily find the problem.

marcus9384, 2018-02-13:
rosdep version: 0.12.1

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 386, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 395, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 592, in command_update
    error_handler=update_error_handler)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 490, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/__init__.py", line 87, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/platforms/debian.py", line 53, in register_platforms
    register_linaro(context)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/platforms/debian.py", line 70, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/local/lib/python2.7/dist-packages/rospkg/os_detect.py", line 656, in get_name
    self.detect_os()
  File "/usr/local/lib/python2.7/dist-packages/rospkg/os_detect.py", line 615, in detect_os
    if os_detector.is_os():
  File "/usr/local/lib/python2.7/dist-packages/rospkg/os_detect.py", line 368, in is_os
    return std_out.strip() == self._os_name_qnx
AttributeError: 'NoneType' object has no attribute 'strip'

mikaelarguedas, 2018-02-15:
This looks to be an issue in [rospkg](https://github.com/ros-infrastructure/rospkg/blob/95d2d78499125989ee4ef0a83ac42bb57f619f68/src/rospkg/os_detect.py#L368). maybe this should be ticketed on https://github.com/ros-infrastructure/rospkg

dirk-thomas, 2018-02-15:
@marcus9384 What platform are you experiencing this on?

dirk-thomas, 2018-02-15:
@marcus9384 What platform are you experiencing this on? I assume `QNX` based on the line number. Can you try to invoke `/bin/uname` manually.