ros-infrastructure/rosdep #563: bad installer puzzling

carladiana, 2017-11-06:
I'm on a Mac Powerbook with OS 10.10.5 I've gotten to the point in the install that recommends the ROS dependencies. When I run this line:

> rosdep install --from-paths src --ignore-src --rosdistro kinetic -y

I get this error:

> ERROR: Rosdep experienced an internal error.
> Please go to the rosdep page [1] and file a bug report with the message below.
> [1] : http://www.ros.org/wiki/rosdep
> 
> rosdep version: 0.11.8
> 
> Bad installer [pip]: [Errno 2] No such file or directory

I reported this issue earlier and the advice was to install pip, so I tried to reinstall it and got this confirmation:

> Requirement already up-to-date: pip in /usr/local/lib/python2.7/site-packages

I'm not sure what to attempt next. Should pop be in a different directory? Or a different version of pip?

NikolausDemmel, 2017-11-06:
rosdep cannot find the `pip` executable. Maybe it is not on your `PATH`? What happens when you run pip from command line?

```
pip --version
```

carladiana, 2017-11-06:
Here's what I get:
-bash: pip: command not found

I'm trying to get a better understanding of Homebrew and Python, so I am looking [here](https://docs.brew.sh/Homebrew-and-Python.html), but open to any advice!  Thanks so much.

NikolausDemmel, 2017-11-06:
If you are using homebrew python, the executable is called `pip2`, not pip. You will need to make sure rosdep can find it as `pip`, e.g. by setting up a shim like https://github.com/mikepurvis/ros-install-osx/blob/master/shim/pip 

If you are new to ROS on OS X, you might want to try Mike Purvis' install script (https://github.com/mikepurvis/ros-install-osx). It takes care of this issue and much more. However, it seems it is geared towards 10.11 and newer, so it might not work as is.

One fundamental issue is that homebrew is a moving target and in particular their python setup has been changing a low, so what once worked might not work any more...

carladiana, 2017-11-06:
Thank you so much. I will try both suggestions.

carladiana, 2017-11-11:
I've decided that for the moment the OS X installation may be too challenging and I will try ROS on another platform. Thank you for your help.

NikolausDemmel, 2017-11-12:
Sure. I'll close the issue then. Best wishes.

closed 2017-11-12 by NikolausDemmel