ros-infrastructure/rosdep #565: Package version ":{version}" does not follow version conventions

kamalpandey1993, 2017-12-22:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 137, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 383, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 436, in _package_args_handler
    pkgs = find_catkin_packages_in(path, options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/catkin_packages.py", line 33, in find_catkin_packages_in
    packages = find_packages(path)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 86, in find_packages
    packages = find_packages_allowing_duplicates(basepath, exclude_paths=exclude_paths, exclude_subspaces=exclude_subspaces, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 146, in find_packages_allowing_duplicates
    xml, filename=filename, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/package.py", line 594, in parse_package_string
    pkg.validate(warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/package.py", line 268, in validate
    raise InvalidPackage('\n'.join(errors))
InvalidPackage: Package version ":{version}" does not follow version conventions

tfoote, 2017-12-23:
Please provide instructions for how to reproduce this issue. It looks like you might have an invalid version defined in one of your packages.

kamalpandey1993, 2018-01-02:
yep got it. the problem was that i was cloning the ros packages from github to the /src folder of ros , this may have been creating the error ...

closed 2018-01-02 by kamalpandey1993