ros-infrastructure/rosdep #570: Matplotlib not installable with Homebrew on Mac

kdhansen, 2018-01-11:
I am running [Mike Purvis' install script for Mac] (https://github.com/mikepurvis/ros-install-osx) and Rosdep runs into an erro when installing matplotlib. It seems that Matplotlib is no longer installable from Homebrew. (Or maybe I missed something)

Output from Rosdep:
```
rosdep version: 0.11.8

Bad installer [homebrew]: Error while parsing brew info for 'homebrew/science/matplotlib'
 * Output of `brew info homebrew/science/matplotlib --json=v1`:

 * Error while parsing:
 Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/rosdep2/platforms/osx.py", line 198, in is_installed
    pkg_info = json.loads(std_out)
  File "/usr/local/Cellar/python/2.7.14_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/local/Cellar/python/2.7.14_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/Cellar/python/2.7.14_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
```

wjwwood, 2018-01-11:
The problem is that output of `brew info homebrew/science/matplotlib --json=v1` is empty (or that's what it looks like). Can you run that command yourself and make sure you have the latest version of Homebrew installed?

kdhansen, 2018-01-11:
My Homebrew is at the latest version. The output is:
```
kdh@h85:lunar_desktop_full_ws$ brew info homebrew/science/matplotlib --json=v1
Error: No available formula with the name "homebrew/science/matplotlib"
```

kdhansen, 2018-01-11:
It looks like the Homebrew policy is that Matplotlib should be "vendored" into the relevant applications. https://github.com/Homebrew/homebrew-core/issues/22710

I guess it should be installed through pip then?

wjwwood, 2018-01-11:
Looks like there is no way to install matplotlib through Homebrew anymore, so the rosdep keys need to updated or something. I don't have time to deal with that right now, but maybe someone else does. You might ask if anyone else has had time to find a fix in the https://github.com/mikepurvis/ros-install-osx repository.

wjwwood, 2018-01-11:
Either way, this isn't an issue with rosdep, it's a problem with the rosdep keys, so I'll close this.

closed 2018-01-11 by wjwwood

wjwwood, 2018-01-11:
Can you try `brew info brewsci/science/matplotlib --json=v1`?

kdhansen, 2018-01-12:
The problem is a local rosdep.yaml in https://github.com/mikepurvis/ros-install-osx with the wrong key.

wjwwood, 2018-01-12:
@kdhansen you're right, the one in rosdistro installs matplotlib from pip. You might need to open an issue against the script then.