ros-infrastructure/rosdep #575: rosdeb with orocos 2.9 workspace

A-Gaber, 2018-01-16:
I am trying to install kinetic and orocos 2.9 but I always get this 
ERROR: Rosdep experienced an error: Invalid manifest file: depends is missing 'package' attribute
rosdep version: 0.11.8

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 137, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 383, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 477, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 665, in command_install
    uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 440, in get_uninstalled
    resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 383, in resolve_all
    rosdep_keys = self.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 305, in get_rosdeps
    return self.loader.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 137, in get_rosdeps
    if resource_name in self.get_catkin_paths():
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 126, in get_catkin_paths
    self._catkin_packages_cache = dict(find_catkin_paths(self._rospack))
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 125, in find_catkin_paths
    filter(lambda x: src.get_manifest(x).is_catkin, src.list()))
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 125, in <lambda>
    filter(lambda x: src.get_manifest(x).is_catkin, src.list()))
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 167, in get_manifest
    return self._load_manifest(name)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 211, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 434, in parse_manifest_file
    return parse_manifest(manifest_name, f.read(), filename)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 470, in parse_manifest
    m.depends = _check_depends(type_, p, filename)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 133, in _check_depends
    raise InvalidManifest("Invalid manifest file [%s]: depends is missing '%s' attribute" % (filename, type_))
InvalidManifest: Invalid manifest file [/home/me/LAB/orocos-2.9_ws/src/orocos_toolchain/orogen/manifest.xml]: depends is missing 'package' attribute

wjwwood, 2018-01-16:
Well, what's the content of `/home/me/LAB/orocos-2.9_ws/src/orocos_toolchain/orogen/manifest.xml`? Is it a valid ROS `manifest.xml` (http://wiki.ros.org/rosbuild/Manifest)?

jackchenxiaojiao, 2018-04-12:
Same problem. Any solutions?

ERROR: Rosdep experienced an error: Invalid manifest file [/home/yii/isir/orocos-2.9_ws/src/orocos_toolchain/orogen/manifest.xml]: depends is missing 'package' attribute
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.12.2

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 484, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 682, in command_install
    uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 445, in get_uninstalled
    resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 393, in resolve_all
    rosdep_keys = self.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 315, in get_rosdeps
    return self.loader.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 138, in get_rosdeps
    if resource_name in self.get_catkin_paths():
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 127, in get_catkin_paths
    self._catkin_packages_cache = dict(find_catkin_paths(self._rospack))
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 126, in find_catkin_paths
    filter(lambda x: src.get_manifest(x).is_catkin, src.list()))
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 126, in <lambda>
    filter(lambda x: src.get_manifest(x).is_catkin, src.list()))
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 167, in get_manifest
    return self._load_manifest(name)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 211, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 434, in parse_manifest_file
    return parse_manifest(manifest_name, f.read(), filename)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 470, in parse_manifest
    m.depends = _check_depends(type_, p, filename)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 133, in _check_depends
    raise InvalidManifest("Invalid manifest file [%s]: depends is missing '%s' attribute" % (filename, type_))
InvalidManifest: Invalid manifest file [/home/yii/isir/orocos-2.9_ws/src/orocos_toolchain/orogen/manifest.xml]: depends is missing 'package' attribute

NikolausDemmel, 2018-04-12:
@jackchenxiaojiao, what about William's answer directly above your comment?

jackchenxiaojiao, 2018-04-12:
@NikolausDemmel . Thank you for your reply.
The content of the /orocos-2.9_ws/src/orocos_toolchain/orogen/manifest.xml file is listed below. 
Sorry I am new to ROS, and I can't find any problem compared to a official ROS XML file as William provided. 

<package>
    <description brief="Orocos/RTT component specification and code generation tool">
        orogen offers a specification interface for components developped in the
        Orocos Realtime Toolkit. Moreover, it generates the specified modules,
        thus offering an easy and fast development process.
    </description>
    <maintainer>Sylvain Joyeux/sylvain.joyeux@m4x.org</maintainer>
    <copyright>
        DFKI/robotik@dfki.de
        Sylvain Joyeux/sylvain.joyeux@m4x.org
    </copyright>
    <license>GPL v2 or later</license>
    <url>http://rock-robotics.org/documentation/orogen</url>

    <depend package="rtt" />
    <depend package="typelib" />
    <depend package="rtt_typelib" />
    <depend package="utilrb" />
    <depend package="metaruby" />
    <depend_optional package="hoe-yard" />
    <depend_optional package="kramdown" />
    <depend name="rake" />
    <depend name="ruby" />

    <test_depend package="minitest" />
    <test_depend package="flexmock" />
</package>

NikolausDemmel, 2018-04-12:
@jackchenxiaojiao, you can use triple-backtricks ``` for code blocks. I updated your comment.

NikolausDemmel, 2018-04-12:
Looks like instead of `name=` it should be `package=`, see https://github.com/orocos-toolchain/orogen/pull/111.

jackchenxiaojiao, 2018-04-12:
Yes, It solves the problem. Thanks very much! @NikolausDemmel

closed 2018-04-12 by NikolausDemmel