ros-infrastructure/rosdep #576: rosdep update ScannerError

pradhanadvait, 2018-01-16:
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.8

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 137, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 381, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 389, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 579, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 472, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 146, in get_gbprepo_as_rosdep_data
    distro_file = get_release_file(gbpdistro)
  File "/usr/lib/python2.7/dist-packages/rosdep2/rosdistrohelper.py", line 76, in get_release_file
    dist_file = rosdistro.get_distribution_file(get_index(), distro)
  File "/usr/lib/python2.7/dist-packages/rosdistro/__init__.py", line 116, in get_distribution_file
    data = _get_dist_file_data(index, dist_name, 'distribution')
  File "/usr/lib/python2.7/dist-packages/rosdistro/__init__.py", line 194, in _get_dist_file_data
    data.append(_load_yaml_data(u))
  File "/usr/lib/python2.7/dist-packages/rosdistro/__init__.py", line 187, in _load_yaml_data
    return yaml.load(yaml_str)
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
    return loader.get_single_data()
  File "/usr/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 170, in fetch_more_tokens
    return self.fetch_stream_end()
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 374, in fetch_stream_end
    self.remove_possible_simple_key()
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 316, in remove_possible_simple_key
    "could not found expected ':'", self.get_mark())
ScannerError: while scanning a simple key
  in "<string>", line 4840, column 3:
      urdf
      ^
could not found expected ':'
  in "<string>", line 4840, column 7:
      urdf
          ^

wjwwood, 2018-01-17:
This looks like a problem where the files are unexpectedly truncated. Can you try again, perhaps it's a network issue?

wjwwood, 2018-01-17:
I'm going to close this because I believe the issue has been addressed. Feel free to continue to comment and I can reopen if necessary.

closed 2018-01-17 by wjwwood

pradhanadvait, 2018-01-18:
Sir, I have tried more than once, but the same problem occurs everytime.

wjwwood, 2018-01-18:
Are you able to do `curl https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml` on the same machine?

It looks like SSL is timing out, it looks like a network issue to me.

pradhanadvait, 2018-01-18:
Ok, I tried it and it gives a perfect result.
Now that I see it more carefully, it says it was unable to process a line which read:
ubuntu: [unrar]
Is that supposed to happen?

pradhanadvait, 2018-01-18:
Will most probably be solved by installing Ubuntu with GPT partition.
Reference: https://github.com/ros/rosdistro/issues/16585

tfoote, 2018-01-18:
@pradhanadvait That does **not** sound like a likely fix for this problem. You're clearly having an ssl connection issue. `<urlopen error ('_ssl.c:574: The handshake operation timed out',)>` Do you have your system time correctly set? I would recommend trying to use the python API to pull from the github urls over https.

pradhanadvait, 2018-01-20:
@tfoote I have no idea on how to do that... can you share some guide about it?

tfoote, 2018-01-22:
There are a lot of similar questions already asked relating to this. Please take a look on answers.ros.org. I searched for `site:answers.ros.org rosdep ssl unable to process timed out` and found quite a few relevant results. https://www.google.com/search?q=site%3Aanswers.ros.org+rosdep+ssl+unable+to+process+timed+out

Lyoid, 2018-07-03:
i run the command "rosdep update" several times and finish the update at last.because i found different files not be hitted during the several times update,i think it is a network problem.
hopes to help you.

KingCoCos, 2018-07-23:
Have you solved this problem?I have encountered the same problem now.

hosh0425, 2019-08-19:
If you are working on embedded boards, it is possible that the OS time is not set correctly. It will cause problems in updating ssh key and also on executing  `sudo rosdep init` &` rosdep update`.