ros-infrastructure/rosdep #577: Cannot run rosdep update. Errno 20: Not a directory

tfburns, 2018-01-17:
I am following [this tutorial](http://wiki.ros.org/kinetic/Installation/Ubuntu), choosing ros-kinetic-desktop-full in step 1.4.

I am able to run "sudo rosdep init" without problem, and this creates the "/etc/ros/rosdep/sources.list.d/20-default.list" file. However, when I then run "rosdep update" I get the following error output:

`reading in sources list data from /etc/ros/rosdep/sources.list.d

ERROR: Rosdep experienced an error: [Errno 20] Not a directory: '/home/tom/.ros/rosdep'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.8

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 137, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 381, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 389, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 579, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 459, in update_sources_list
    retval.append((source, write_cache_file(sources_cache_dir, source.url, rosdep_data)))
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 537, in write_cache_file
    os.makedirs(source_cache_d)
  File "/usr/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 20] Not a directory: '/home/tom/.ros/rosdep'
`

Running "rosdep check" returns:

`ERROR: your rosdep installation has not been initialized yet.  Please run:
        
    rosdep update
`

This is on a fresh installation of Xenial (Ubuntu 16.04).

wjwwood, 2018-01-17:
Does `/home/tom/.ros/rosdep` exist? Is it perhaps a file and not a directory?

tfburns, 2018-01-17:
'.ros' exists as a file in '/home/tom/' and has the value "1".

tfburns, 2018-01-17:
I deleted the ".ros" file and created to new folders so the directory "/home/tom/.ros/rosdep" exists and got to the end of the above-mentioned tutorial without further errors.