ros-infrastructure/rosdep #582: ROS_ETC_DIR inconsistency

paulbovbel, 2018-02-08:
It appears that this PR (https://github.com/ros-infrastructure/rosdep/pull/551) has introduced an issue where `sudo rosdep init` will write to `/etc/ros/rosdep/sources.list.d`, while `rosdep update` will try to read from whichever environment is sourced, typically `/opt/ros/$ROS_DISTRO/etc/ros/rosdep/sources.list.d`.

dirk-thomas, 2018-02-08:
@paulbovbel Can you please provide more detailed information: what platform are you on, what are your environment variables set to.

@allenh1 Since you created the referenced patch can you please look into this.

allenh1, 2018-02-08:
@dirk-thomas I can certainly do that -- will give it a go in a container in a little bit and let you know what I get.

paulbovbel, 2018-02-08:
Reproduction on a clean xenial container:

```
# preliminary
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list &&
sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116 &&
sudo apt-get update &&
sudo apt-get install ros-kinetic-ros-base python-pip

# install rosdep 0.12.0
sudo pip install rosdep

# rosdep update fails
source /opt/ros/kinetic/setup.bash &&
sudo rosdep init &&
rosdep update
```

allenh1, 2018-02-08:
Okay, I found something after looking into this.

```
root@c1cf676de24c:~# echo ${ROS_ETC_DIR}

root@c1cf676de24c:~# source /opt/ros/kinetic/setup.bash 
root@c1cf676de24c:~# echo ${ROS_ETC_DIR}
/opt/ros/kinetic/etc/ros
```

The environment variable gets set by `/opt/ros/kinetic/etc/catkin/profile.d/10.ros.sh`, which belongs to `ros-kinetic-roslib`.

@dirk-thomas Perhaps a better patch would be to introduce a `ROSDEP_ETC_DIR` instead? Let me know and I'll jump right on it.

dirk-thomas, 2018-02-08:
Since the reference PR breaks the behavior of `rosdep update` when a setup file has been sourced I will go ahead and revert the change and release a new patch version of `rosdep`. @paulbovbel Thank you bringing this up.

@allenh1 Please feel free to create a follow up PR to allow different locations for the rosdep source files. I just don't want to wait for a patch, testing it, let others comment and review to address the regression.

allenh1, 2018-02-08:
> I just don't want to wait for a patch, testing it, let others comment and review to address the regression.

That's a fair point -- this is certainly not good. My apologies, I should have tested sourcing `setup.bash` before running `sudo rosdep init`.

I've got a proposed fix up for it, which I will test far more thoroughly.

dirk-thomas, 2018-02-08:
The referenced change has been reverted in #584 and a new version `0.12.1` has been released.

closed 2018-02-08 by dirk-thomas