ros-infrastructure/rosdep #586: Homebrew issue on macOS High Sierra (10.13.1 17B1003).

lizixroy, 2018-02-11:
Hi folks,

I have been trying to install ROS on my Mac running macOS High Sierra (10.13.1 17B1003)., it's not going well. I know that MacOS is an officially supported OS, but it would be so nice if we can get this to work, because there are so many Mac users interested in leveraging ROS. 

The issue I'm getting is with rosdep, it fails on a homebrew command: 
```
ERROR: the following rosdeps failed to install
  homebrew: command [brew install ros/kinetic/gazebo_ros_pkgs] failed
```

After going through rosdep source code, I found that the way that the name of homebrew formulae are created might not work anymore. In get_gbprepo_as_rosdep_data method in gbpdistro_support.py, name of a formula is created by chaining together tap, release name and rosdep_key:
`homebrew_name = '%s/%s/%s' % (tap, release_name, rosdep_key)`
After that, 'brew install' is called with this homebrew_name: 
`executing command [brew install ros/kinetic/gazebo_ros_pkgs]`

However homebrew doesn't work with "ros/kinetic/gazebo_ros_pkgs":

```
Error: No available formula with the name "ros/kinetic/gazebo_ros_pkgs" 
Please tap it and then try again: brew tap ros/kinetic
executing command [brew install ros/kinetic/gazebo_ros_pkgs]
==> Tapping ros/kinetic
Cloning into '/usr/local/Homebrew/Library/Taps/ros/homebrew-kinetic'...
remote: Repository not found.
fatal: repository 'https://github.com/ros/homebrew-kinetic/' not found
Error: Failure while executing: git clone https://github.com/ros/homebrew-kinetic /usr/local/Homebrew/Library/Taps/ros/homebrew-kinetic --depth=1
ERROR: the following rosdeps failed to install
```

This renders resdep not functioning on macOS.

Any idea why this is happening?

Have you guys stopped supporting homebrew?

It's totally OK if I myself need to take time to make ROS installation on Mac work again, just need some pointers.

Thanks much for your time!

Roy

NikolausDemmel, 2018-02-12:
General remark: The best place to start with ROS installation on OSX is probably: https://github.com/mikepurvis/ros-install-osx

Note there is some open discussion on High Sierra support, e.g. https://github.com/mikepurvis/ros-install-osx/issues/100, https://github.com/mikepurvis/ros-install-osx/issues/86 (I didn't read).

Concerning your issue here with rosdep, looks like the `ros/kinetic/...` formulae is the fallback of rosdep where it assumes that otherwise unspecified dependencies are ros packages. Unfortunately, so far ros packages are not available as homebrew packages at all, i.e. you need to install from source. So I think this issue can be closed? I propose you check out https://github.com/mikepurvis/ros-install-osx and raise any remaining issues there, unless there is indeed a rosdep bug.

PS: Maybe this fallback behaviour should be changed for the time being, such that the error message in that case is more reasonable and doesn't imply that it should be working. A PR that changes the error message would be welcome!

wjwwood, 2018-02-12:
@NikolausDemmel's answer is correct. I'd translate the message:

> homebrew: command [brew install ros/kinetic/gazebo_ros_pkgs] failed

To be something like:
> you have a package which depends on gazebo_ros_pkgs but gazebo_ros_pkgs is neither on your system (i.e. not on the `ROS_PACKAGE_PATH` or found under the `--from-paths` argument) nor installable by homebrew (i.e. `brew install ros/kinetic/gazebo_ros_pkgs` fails)

wjwwood, 2018-02-12:
I'm going to close this because I believe the issue has been addressed. Feel free to continue to comment and I can reopen if necessary.

As @NikolausDemmel said, a pr to improve the error message would be appreciated.

closed 2018-02-12 by wjwwood

nik312123, 2018-10-04:
Hi! This issue still exists for me even with the mikepurvis/ros-install-osx when building packages from a repository for my robotics team. I cloned their repo, and when I try to install the project from source using rosdep install --from-paths src --ignore-src -r -y --os=osx:high\ sierra, this issue occurs for me.