ros-infrastructure/rosdep #587: Ability to query build or exec dependencies only

koenlek, 2018-02-25:
In package xml format v2 the following dependency types exist ([quoted from the official wiki](http://wiki.ros.org/action/fullsearch/catkin/package.xml)):

* `<depend>` (specifies that a dependency is a build, export, and execution dependency. This is the most commonly used dependency tag)
* `<buildtool_depend>`
* `<build_depend>`
* `<build_export_depend>`
* `<exec_depend>`
* `<test_depend>`
* `<doc_depend>`

In format v1, a similar division in dependency types can be made. 

The problem we run into is that `rosdep` seems unable to differentiate between the different dependency types. E.g. `rosdep keys` will return keys for all dependency types (except `doc_depend`, based on a quick test I just did). 

For packaging and deployment, we would like to be able to differentiate between runtime (all `exec_depend` + `depend` keys) dependencies and the "export build" dependencies (i.e. for packages that want to build against this package), which would consist of all `build_export_depend` and `depend` keys. 

Currently, I've made a workaround solution which temporarily removes the irrelevant lines from the `package.xml` with a `sed` command and then runs the `rosdep` command. I wonder if other people agree that support within the `rosdep` tool to select different dependency types would make more sense. I am also curious how bloom actually solves this: bloom needs to separate run dependencies from build dependencies too, right? 

I quickly checked out the rosdep code to see how it actually implements the package.xml parsing, but I couldn't find it yet. I couldn't even find `exec_depend` in the rosdep code, so I guess the parsing logic might even be in an external repository/tool? If people are willing to think with me how it would need to be implemented, I might be able to provide a PR for it.

What about a option like `--filter-dependency-types depend exec_depend`? This way, providing support for both v1 and v2 seems straightforward, as people can just give a list of the xml element names they want to filter.

nalt, 2019-10-28:
See https://github.com/ros-infrastructure/rosdep/pull/727

Aposhian, 2021-07-20:
This should be closed by #789

nuclearsandwich, 2021-07-20:
> This should be closed by #789

Yes indeed!

closed 2021-07-20 by nuclearsandwich