ros-infrastructure/rosdep #589: ERROR: Rosdep experienced an error: list index out of range in debian stretch (installation from source)

kamalpandey1993, 2018-03-01:
rosdep version: 0.12.1

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 484, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 706, in command_install
    installer.install(uninstalled, **install_options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 525, in install
    verbose=verbose, quiet=quiet)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 588, in install_resolved
    if isinstance(sub_command[0], list):  # list of alternatives
IndexError: list index out of range
while installing ros on my beagleboneblack(armhf) with debian stretch, from source this issue was found

wjwwood, 2018-03-01:
Can you please tell us what rosdep command you ran exactly and also rerun it with `--simulate` on and provide the complete console output, not just the error. Thanks.

kamalpandey1993, 2018-03-04:
` rosdep install --from-paths src --ignore-src --rosdistro lunar -y ` was the command 
this is the output after putting `--simulate`
`#[apt] Installation commands:

ERROR: Rosdep experienced an error: list index out of range
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.12.1

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 484, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 706, in command_install
    installer.install(uninstalled, **install_options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 525, in install
    verbose=verbose, quiet=quiet)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 564, in install_resolved
    if isinstance(sub_command[0], list):
IndexError: list index out of range

`

flixr, 2018-03-12:
I'm getting the same error on xenial/kinetic:
```
$ rosdep install --from-paths . --ignore-src -r -y
Reading package lists...
Building dependency tree...
Reading state information...
The following NEW packages will be installed:
  ros-kinetic-xacro
0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.
Need to get 40.9 kB of archives.
After this operation, 155 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-xacro amd64 1.11.2-0xenial-20171117-013356-0800 [40.9 kB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 40.9 kB in 0s (0 B/s)
Selecting previously unselected package ros-kinetic-xacro.
...
(Reading database ... 87945 files and directories currently installed.)
Preparing to unpack .../ros-kinetic-xacro_1.11.2-0xenial-20171117-013356-0800_amd64.deb ...
Unpacking ros-kinetic-xacro (1.11.2-0xenial-20171117-013356-0800) ...
Setting up ros-kinetic-xacro (1.11.2-0xenial-20171117-013356-0800) ...

ERROR: Rosdep experienced an error: list index out of range
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.12.1

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 484, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 706, in command_install
    installer.install(uninstalled, **install_options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 525, in install
    verbose=verbose, quiet=quiet)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 588, in install_resolved
    if isinstance(sub_command[0], list):  # list of alternatives
IndexError: list index out of range

executing command [apt-get install -y ros-kinetic-xacro]
```

flixr, 2018-03-12:
Seems this regression was introduced in #521

ipa-mdl, 2018-03-12:
@flixr: I cannot reproduce this error with a test package that depends on `xacro`.
Can you provide me an example workspace, please?

ArcEye, 2018-03-21:
Exactly the same here on a fresh install of Stretch

```
Command:
$rosdep install --from-paths src --ignore-src --rosdistro kinetic -y -r --os=debian:stretch

Error Output:

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rrbot_control: No definition of [effort_controllers] for OS version [stretch]
simulators: No definition of [stage_ros] for OS version [stretch]
rrbot_gazebo: No definition of [gazebo_ros_control] for OS version [stretch]
Continuing to install resolvable dependencies...

ERROR: Rosdep experienced an error: list index out of range
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.12.1

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 484, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 706, in command_install
    installer.install(uninstalled, **install_options)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/installers.py", line 525, in install
    verbose=verbose, quiet=quiet)
  File "/usr/local/lib/python2.7/dist-packages/rosdep2/installers.py", line 588, in install_resolved
    if isinstance(sub_command[0], list):  # list of alternatives
IndexError: list index out of range
```

flixr, 2018-03-21:
@ArcEye should be fixed when #585 is merged.

wjwwood, 2018-03-21:
Closed by https://github.com/ros-infrastructure/rosdep/pull/585 and `0.12.2` is being released now with that fix.

closed 2018-03-21 by wjwwood