ros-infrastructure/rosdep #590: ERROR: Rosdep experienced an error: list index out of range

MingCHEN-Github, 2018-03-02:
ERROR: Rosdep experienced an error: list index out of range
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.12.1

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 484, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 706, in command_install
    installer.install(uninstalled, **install_options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 525, in install
    verbose=verbose, quiet=quiet)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 588, in install_resolved
    if isinstance(sub_command[0], list):  # list of alternatives
IndexError: list index out of range

flixr, 2018-03-12:
Seems to be a duplicate of #589

wjwwood, 2018-03-12:
Indeed, I'll try to get to the bottom of this issue asap.

closed 2018-03-12 by wjwwood

flixr, 2018-03-12:
@wjwwood https://github.com/ros-infrastructure/rosdep/pull/592 should fix it...
At least my local builds work again...

wjwwood, 2018-03-12:
Yeah, sorry I was still catching up in chronological order, thanks for taking a stab at it.