ros-infrastructure/rosdep #596: flake8 executed in 2 different ways with different settings

Nickolaim, 2018-03-17:
rosdep has flake8 unit test (see [https://github.com/ros-infrastructure/rosdep/blob/master/test/test_flake8.py](https://github.com/ros-infrastructure/rosdep/blob/master/test/test_flake8.py)]

Pull Request  #593 also run flake8 but in Travis CI with a different set of options.

What's the value of having flake8 being run in 2 different ways?  Running them from unit test seems like the best option - unit tests not only executed in Travis CI but on local dev machines.

@cclauss , @wjwwood - you might have an opinion.

I am proposing to use one approach and merge the rules; I personally prefer the unit test.  Happy to do the change.

cclauss, 2018-03-17:
I have no objection to this course of action.  I agree that running flake8 in two different locations is suboptimal.

wjwwood, 2018-03-20:
Oh, I didn't notice (we have a flake8 automated test on some repositories but not others.

I have a slight preference for the automated test, rather than running it explicitly in travis, simply because then that test is run locally when running tests too.

Pull requests to unify this would be appreciated.

Nickolaim, 2018-03-20:
I have a fix for this.  Will do a PR later today.

Sent from a mobile device.

closed 2018-03-23 by Nickolaim