ros-infrastructure/rosdep #597: ERROR: Rosdep experienced an error: Package version ":{version}" does not follow version conventions

jannsta1, 2018-03-19:
I got the following traceback while using rosdep and as requested by the warning message I am submitting a bug report with the stack trace below:

```
edison@jubilinux:~/ros_catkin_ws$ rosdep install --from-paths src --ignore-src -y

ERROR: Rosdep experienced an error: Package version ":{version}" does not follow version conventions
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.12.1

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 443, in _package_args_handler
    pkgs = find_catkin_packages_in(path, options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/catkin_packages.py", line 33, in find_catkin_packages_in
    packages = find_packages(path)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 86, in find_packages
    packages = find_packages_allowing_duplicates(basepath, exclude_paths=exclude_paths, exclude_subspaces=exclude_subspaces, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 146, in find_packages_allowing_duplicates
    xml, filename=filename, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/package.py", line 664, in parse_package_string
    pkg.validate(warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/package.py", line 305, in validate
    raise InvalidPackage('\n'.join(errors))
InvalidPackage: Package version ":{version}" does not follow version conventions
```

wjwwood, 2018-03-19:
One of the `package.xml`'s in your workspace has an invalid version value. I recommend filing an issue against the package which has this.

closed 2018-03-19 by wjwwood

wjwwood, 2018-03-19:
At the very least, please let us know on this issue which `package.xml` in your `src` folder contains `:{version}`, and I can help point you to the right place to report the issue.

jannsta1, 2018-03-19:
OK, thanks @wjwwood , the package was mavlink-gbp-release. 

**NB.** I probably wouldn't have reported this usually but the traceback read like bug reports are desirable if this traceback is observed, at least that was my interpretation.

wjwwood, 2018-03-19:
> OK, thanks @wjwwood , the package was mavlink-gbp-release.

So that might be because of where you're getting the `package.xml` or it could be that the maintainer of that package didn't run `bloom` correctly, because `bloom` is supposed to replace the `:{version}` with an actual version.

> I probably wouldn't have reported this usually but the traceback read like bug reports are desirable if this traceback is observed, at least that was my interpretation.

Yeah, we could probably improve the error message in this case, though I don't have time to do that right now. I'll open an issue to track that specifically.