ros-infrastructure/rosdep #598: improve error message when package.xml is malformed

wjwwood, 2018-03-19:
Anytime there's an issue with a `package.xml` you get something like reported in https://github.com/ros-infrastructure/rosdep/issues/597#issue-306498319.

Instead it should avoid the traceback and the message to report it on the rosdep repository. Also, it should include which `package.xml` has the issue.

Nickolaim, 2018-03-20:
I am happy to work on this but I need a repro that I can do on a clean machine.  I suspect in this case it should include invalid package.xml and a command line.

wjwwood, 2018-03-20:
I think you could take any workspace (like with just catkin in it) and modify the package.xml to have `:{version}` in the `<version>` tag. I'm not sure about the other failure modes. Might need to read the logic in catkin_pkg to see what errors can be thrown.

Nickolaim, 2018-03-21:
So here is my plan: 
1) Update `catkin_pkg` repo, so that the error message for `InvalidPackage` exception contains the file name if it is available
2) Update the main exception handler in `rosdep` so it treats `InvalidPackage` specially - just print the error message and exits (no traceback, no standard exception message)
3) Add tests to both repos

How does it sound?

Nickolaim, 2018-05-22:
All changes are in, I propose to close the issue.

closed 2018-05-22 by wjwwood