ros-infrastructure/rosdep #610: 'rosdep update' raises exception when run on Windows

jonbonne, 2018-05-25:
Windows 10 Education edition
The exception was raised while running rosdep update on Windows after installing the ROS bootstrap utilities via pip. Was able to fix the problem locally by catching the exception and setting the 'self.sudo_command' attribute indicated at the end of the stack track to ''. rosdep udpate seems to run without issue now.

rosdep version: 0.12.2

Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\rosdep2\main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "C:\Python27\lib\site-packages\rosdep2\main.py", line 386, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "C:\Python27\lib\site-packages\rosdep2\main.py", line 395, in _no_args_handler
    return command_handlers[command](options)
  File "C:\Python27\lib\site-packages\rosdep2\main.py", line 592, in command_update
    error_handler=update_error_handler)
  File "C:\Python27\lib\site-packages\rosdep2\sources_list.py", line 490, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "C:\Python27\lib\site-packages\rosdep2\gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "C:\Python27\lib\site-packages\rosdep2\__init__.py", line 81, in create_default_installer_context
    m.register_installers(context)
  File "C:\Python27\lib\site-packages\rosdep2\platforms\source.py", line 53, in register_installers
    context.set_installer(SOURCE_INSTALLER, SourceInstaller())
  File "C:\Python27\lib\site-packages\rosdep2\platforms\source.py", line 201, in __init__
    super(SourceInstaller, self).__init__(source_detect, supports_depends=True)
  File "C:\Python27\lib\site-packages\rosdep2\installers.py", line 328, in __init__
    self.sudo_command = 'sudo -H' if os.geteuid() != 0 else ''
AttributeError: 'module' object has no attribute 'geteuid'

tfoote, 2018-05-25:
sudo certainly doesn't make sense on Windows. Is there an equivalent detection of being an Administrator? If there's not this sort of patch might make sense. However it would be better to specifically check for Windows, instead of catching a generic AttributeError that could be caused by something else.

There's a command line option to disable using sudo, `--as-root` that I'd recommend using in the mean time. What package manager are you trying to use with rosdep? You'll need to pass it into this argument.

ImanHosseini, 2021-04-11:
I am on [win10] rosdep version 0.20.0, "rosdep update --as-root pip:no" fails with the same 'has no attribute geteuid' error. is there a way around it?