ros-infrastructure/rosdep #614: Bash script to replace package list doesn't work

ejalaa12, 2018-06-27:
As described [here](http://wiki.ros.org/rosdep/rosdep.yaml), it is possible to install system dependency that is not available in the package manager.
I am trying to run the default example

I created a `custom_deps.yaml` file, with the following content:
```yaml
mytest_rosdep_name:
    ubuntu: |
        echo "hello world"
```

And i append this file to `/etc/ros/rosdep/sources.list.d/20-default.list`

```bash
echo file://$(readlink -f custom_deps.yaml) >> /etc/ros/rosdep/sources.list.d/20-default.list
```

After updating rosdep (`rosdep update`), i try the following command to test:
```bash
> rosdep install myRosPkg --simulate -i --os=ubuntu:xenial
rosdep detected OS: [elementary] aliasing it to: [ubuntu]
#[apt] Installation commands:
  sudo -H apt-get install echo
  sudo -H apt-get install "hello
  sudo -H apt-get install world
```

What did i do wrong ?

tfoote, 2018-06-27:
That format is no longer supported. 

The top of that page documents that the format described there is only supported through 0.11

`This page is only applicable to ROS v0.11 and older`

Re: https://answers.ros.org/question/295471/installing-a-custom-dependency-with-rosdep/
Please don't double post questions following our support policy: http://wiki.ros.org/Support

closed 2018-06-27 by tfoote

ejalaa12, 2018-06-27:
Sorry for the double post I'll move to ros.answer :)