ros-infrastructure/rosdep #615: rosdep update failed with Mojave OS X

wdestin, 2018-06-28:
```
$ ros_catkin_ws rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"

ERROR: Rosdep experienced an error: unrecognized version: 10.14
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.12.2

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 386, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 395, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 592, in command_update
    error_handler=update_error_handler)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/sources_list.py", line 490, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/local/lib/python2.7/site-packages/rosdep2/__init__.py", line 87, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/platforms/debian.py", line 53, in register_platforms
    register_linaro(context)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/platforms/debian.py", line 70, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/local/lib/python2.7/site-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/local/lib/python2.7/site-packages/rospkg/os_detect.py", line 656, in get_name
    self.detect_os()
  File "/usr/local/lib/python2.7/site-packages/rospkg/os_detect.py", line 618, in detect_os
    self._os_codename = os_detector.get_codename()
  File "/usr/local/lib/python2.7/site-packages/rospkg/os_detect.py", line 343, in get_codename
    return _osx_codename(*ver[0:2])
  File "/usr/local/lib/python2.7/site-packages/rospkg/os_detect.py", line 321, in _osx_codename
    raise OsNotDetected("unrecognized version: %s.%s" % (major, minor))
OsNotDetected: unrecognized version: 10.14
```

tfoote, 2018-06-28:
rospkg.os_detect needs to be extended to support mojave https://github.com/ros-infrastructure/rospkg/blob/f187910cd7af7e34bbe1d6f386bbc27a9f2cbf01/src/rospkg/os_detect.py#L308-L319

nuclearsandwich, 2018-09-30:
I'm going to close this since https://github.com/ros-infrastructure/rospkg/pull/147 was released in rospkg 1.1.5 and mojave detection should be working. Please re-open or leave a comment if that is not the case.

closed 2018-09-30 by nuclearsandwich