ros-infrastructure/rosdep #617: UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 213: ordinal not in range(128

RashikShahjahan, 2018-07-27:
I tried following 2.1.2 of the of this tutorial: http://wiki.ros.org/kinetic/Installation/Source
I got the following error:
rosdep version: 0.12.2
[
Traceback (most recent call last):
  File "/home/rashik/ros/lib/python3.6/site-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/home/rashik/ros/lib/python3.6/site-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/home/rashik/ros/lib/python3.6/site-packages/rosdep2/main.py", line 484, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/home/rashik/ros/lib/python3.6/site-packages/rosdep2/main.py", line 682, in command_install
    uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
  File "/home/rashik/ros/lib/python3.6/site-packages/rosdep2/installers.py", line 445, in get_uninstalled
    resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
  File "/home/rashik/ros/lib/python3.6/site-packages/rosdep2/lookup.py", line 393, in resolve_all
    rosdep_keys = self.get_rosdeps(resource_name, implicit=implicit)
  File "/home/rashik/ros/lib/python3.6/site-packages/rosdep2/lookup.py", line 315, in get_rosdeps
    return self.loader.get_rosdeps(resource_name, implicit=implicit)
  File "/home/rashik/ros/lib/python3.6/site-packages/rosdep2/rospkg_loader.py", line 138, in get_rosdeps
    if resource_name in self.get_catkin_paths():
  File "/home/rashik/ros/lib/python3.6/site-packages/rosdep2/rospkg_loader.py", line 127, in get_catkin_paths
    self._catkin_packages_cache = dict(find_catkin_paths(self._rospack))
  File "/home/rashik/ros/lib/python3.6/site-packages/rosdep2/rospkg_loader.py", line 126, in <lambda>
    filter(lambda x: src.get_manifest(x).is_catkin, src.list()))
  File "/home/rashik/ros/lib/python3.6/site-packages/rospkg/rospack.py", line 167, in get_manifest
    return self._load_manifest(name)
  File "/home/rashik/ros/lib/python3.6/site-packages/rospkg/rospack.py", line 211, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/home/rashik/ros/lib/python3.6/site-packages/rospkg/manifest.py", line 434, in parse_manifest_file
    return parse_manifest(manifest_name, f.read(), filename)
  File "/home/rashik/ros/lib/python3.6/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 213: ordinal not in range(128)
](url)

wjwwood, 2018-07-27:
Looks like you have a `package.xml` or `manifest.xml` file in your workspace which is malformed or maybe which has unicode characters that `rosdep` can't handle. Check each of them and make sure they look appropriate.

Also, please include the output of `rosdep` before the error (all of the output actually).

RashikShahjahan, 2018-07-27:
@wjwwood This was the output before the error.

 ERROR: Rosdep experienced an error: 'ascii' codec can't decode byte 0xc3 in position 213: ordinal not in range(128)
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

RashikShahjahan, 2018-07-27:
@wjwwood  And my catkin workspace is empty if that's what you meant.

dehavenm, 2018-08-08:
I had the same error. It turned out to be the file located at /opt/ros/kinetic/share/roslisp/manifest.xml

In the line with the <author> tag, one of the authors is listed as "Lorenz Mösenlechner" - the umlaut in "Mösenlechner" was the character which was causing the error. Removing it allowed rosdep to work properly.

Though what exactly is the proper procedure for reporting bugs for rosdep? The "Bugs/Feature Requests" part of  http://www.ros.org/wiki/rosdep just points back to this Github page.

RashikShahjahan, 2018-08-08:
@dehavenm I stopped getting the error after running it in a virtualenv containing only python 2. I eventually gave up on it as I got other errors. Now I am trying to set it up on a ubuntu lxc. This[](https://gist.github.com/lcpz/0ab75aa5205504ced9f5c11cac10a89e)

rsinnet, 2018-08-09:
Not sure exactly what happened here but if you remove the `\xf6` character (o with umlaut ) from the `roslisp` `manifest.xml` file, it works. Was there some recent change to `rosdep` that led to this error?

dirk-thomas, 2018-08-09:
Duplicate of ros-infrastructure/rospkg#154.

closed 2018-08-09 by dirk-thomas