ros-infrastructure/rosdep #618: Make rosdep python2 and python3 side by side installable

mikaelarguedas, 2018-08-08:
Something similar to https://github.com/ros-infrastructure/catkin_pkg/pull/157

This will also allows us to use the python3 version of rosdep for the ROS 2 installation instruction without impacting existing ROS 1 install if any

stonier, 2018-09-30:
+1 for this. Would like to maintain a melodic and a bouncy environment in parallel without having to resort to docker containers.

dirk-thomas, 2019-02-20:
I don't want to argue that this package should not be split like the other Python packages. I certainly do think it would be good. I am just not sure what the exact use case is where this is *required*.

> This will also allows us to use the python3 version of rosdep for the ROS 2 installation instruction without impacting existing ROS 1 install if any

The instructions will mentioned to install `python-rosdep` or `python3-rosdep` in order to make the global executable `rosdep` available. That will still not be possible even if the Python modules would be separated into their own Debian package - only these `-modules` packages will be side-by-side installable.

> Would like to maintain a melodic and a bouncy environment in parallel without having to resort to docker containers.

@stonier Why would you need to install `python-rosdep` as well as `python3-rosdep` for that at the same time?

@JWhitleyAStuff Please provide a concrete use case where you require this.

JWhitleyAStuff, 2019-02-20:
@dirk-thomas - my use case has disappeared. I was confusing other issues (which I have since resolved) with this one.

stonier, 2019-02-20:
Aye, not required. I have the python2 variants of the build tooling working for melodic and crystal. Had other problems with python3 variants, and much is clearer now, but not relevant to this post any longer.

mikaelarguedas, 2019-02-21:
Regarding the installation instructions, we moved to installing `python-rosdep` in the ROS2 instructions to avoid disrupting ROS1 users. starting with ROS Noetic we should be able to advise users to install `python3-rosdep` for ROS1 and ROS2.

So for that use case, resolving this ticket will not solve the issue and there is a path forward without having to split the package

JWhitleyAStuff, 2019-02-21:
@mikaelarguedas - We can probably close this issue. I think this conversation provides enough context and solutions for future questions about the issue.

mikaelarguedas, 2019-02-22:
> We can probably close this issue. I think this conversation provides enough context and solutions for future questions about the issue.

Sounds good to me :+1:

closed 2019-02-22 by mikaelarguedas

mikepurvis, 2019-11-19:
Our issue we're coming up against with this is that `rospack.get_depends_on` is needed for plugin discovery; rospack imports rosdep, and neither one has been split, so you can only have rospack and rosdep for either Python 2 or Python 3 (globally, anyway).

For a practical consequence of this, if you have a built-from-source Python 3 workspace on a machine that's otherwise running the Python 2 tools, that Python 3 workspace won't be able to do things like run find rosbag migration rules. This happens to be a fairly major frustration for us, as we have a migrated-to-Py3 workspace that is now no longer co-installable with the previous Py2 workspace, making support development on the previous release a headache.

FYI @guillaumeautran

dirk-thomas, 2019-11-19:
Please see #731.

mikepurvis, 2019-11-20:
Thanks so much for the prompt service!

mikepurvis, 2019-11-20:
Sorry, there's an error in my post above— `rospack` is an in-bundle package, the other standalone package is `rospkg`, which has already been split up (related discussion in https://github.com/ros-infrastructure/rospkg/issues/103).

PR #731 (2019-11-20): split modules into separate Debian package
Resolves #618.