ros-infrastructure/rosdep #620: ERROR: Rosdep experienced an error: [/home/pi/ros_catkin_ws/src/roslisp/manifest.xml] invalid XML: 'ascii' codec can't encode character u'\xf6' in position 213: ordinal not in range(128)

deeaisa-saini, 2018-08-09:
Below command on trying to install ROS Distribution lunar resulted in below error, can you please check and resolve it ASAP please. Thanks!

rosdep install --from-paths src --ignore-src --rosdistro lunar -y

ERROR: Rosdep experienced an error: [/home/pi/ros_catkin_ws/src/roslisp/manifest.xml] invalid XML: 'ascii' codec can't encode character u'\xf6' in position 213: ordinal not in range(128)

rosdep version: 0.12.2

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 484, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 682, in command_install
    uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 445, in get_uninstalled
    resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 393, in resolve_all
    rosdep_keys = self.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 315, in get_rosdeps
    return self.loader.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 138, in get_rosdeps
    if resource_name in self.get_catkin_paths():
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 127, in get_catkin_paths
    self._catkin_packages_cache = dict(find_catkin_paths(self._rospack))
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 126, in find_catkin_paths
    filter(lambda x: src.get_manifest(x).is_catkin, src.list()))
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 126, in <lambda>
    filter(lambda x: src.get_manifest(x).is_catkin, src.list()))
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 167, in get_manifest
    return self._load_manifest(name)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 211, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 434, in parse_manifest_file
    return parse_manifest(manifest_name, f.read().decode('utf-8'), filename)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 454, in parse_manifest
    raise InvalidManifest("[%s] invalid XML: %s" % (filename, e))
InvalidManifest: [/home/pi/ros_catkin_ws/src/roslisp/manifest.xml] invalid XML: 'ascii' codec can't encode character u'\xf6' in position 213: ordinal not in range(128)

TSRyan, 2018-08-09:
I have the same issue installing Kinetic.

rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y

ERROR: Rosdep experienced an error: [/opt/ros/kinetic/share/roslisp/manifest.xml] invalid XML: 'ascii' codec can't encode character u'\xf6' in position 213: ordinal not in range(128)
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.12.2

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 484, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 682, in command_install
    uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 445, in get_uninstalled
    resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 393, in resolve_all
    rosdep_keys = self.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 315, in get_rosdeps
    return self.loader.get_rosdeps(resource_name, implicit=implicit)
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 138, in get_rosdeps
    if resource_name in self.get_catkin_paths():
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 127, in get_catkin_paths
    self._catkin_packages_cache = dict(find_catkin_paths(self._rospack))
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 126, in find_catkin_paths
    filter(lambda x: src.get_manifest(x).is_catkin, src.list()))
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 126, in <lambda>
    filter(lambda x: src.get_manifest(x).is_catkin, src.list()))
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 167, in get_manifest
    return self._load_manifest(name)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 211, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 434, in parse_manifest_file
    return parse_manifest(manifest_name, f.read().decode('utf-8'), filename)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 454, in parse_manifest
    raise InvalidManifest("[%s] invalid XML: %s" % (filename, e))
InvalidManifest: [/opt/ros/kinetic/share/roslisp/manifest.xml] invalid XML: 'ascii' codec can't encode character u'\xf6' in position 213: ordinal not in range(128)

TSRyan, 2018-08-09:
Duplicate of #617 and #619.

dirk-thomas, 2018-08-09:
Duplicate of ros-infrastructure/rospkg#154.

closed 2018-08-09 by dirk-thomas