ros-infrastructure/rosdep #626: rosdep update, cannot complete, failed with missing attribute "encode"

peterso, 2018-08-23:
ERROR: Rosdep experienced an error: 'list' object has no attribute 'encode'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.10.30

Traceback (most recent call last):
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 126, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 286, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 294, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 468, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/pymodules/python2.7/rosdep2/sources_list.py", line 438, in update_sources_list
    retval.append((rds, write_cache_file(sources_cache_dir, dist['distribution'], rosdep_data)))
  File "/usr/lib/pymodules/python2.7/rosdep2/sources_list.py", line 493, in write_cache_file
    key_hash = compute_filename_hash(filename_key)
  File "/usr/lib/pymodules/python2.7/rosdep2/sources_list.py", line 479, in compute_filename_hash
    sha_hash.update(filename_key.encode())
AttributeError: 'list' object has no attribute 'encode'

nuclearsandwich, 2018-09-30:
Can you try updating to a more recent version of rosdep. Two previous reports of this issue were seemingly resolved with an update. #453 #484

nuclearsandwich, 2018-11-07:
@peterso I'm going to close this on the assumption that updating rosdep will resolve the issue. If it doesn't please feel free to re-open and comment here. Thanks.

closed 2018-11-07 by nuclearsandwich