ros-infrastructure/rosdep #632: ERROR: Rosdep experienced an error: Unable to handle package.xml format version '3'

ClarkTucker, 2018-09-29:
ctucker@dawson:~/ros2_2018$ rosdep check --from-paths src --ignore-src   console_bridge

ERROR: Rosdep experienced an error: Unable to handle package.xml format version '3', please update catkin_pkg (e.g. on Ubuntu/Debian use: sudo apt-get update && sudo apt-get install --only-upgrade python-catkin-pkg)
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.4

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 353, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 406, in _package_args_handler
    pkgs = find_catkin_packages_in(path, options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/catkin_packages.py", line 33, in find_catkin_packages_in
    packages = find_packages(path)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 83, in find_packages
    packages = find_packages_allowing_duplicates(basepath, exclude_paths=exclude_paths, exclude_subspaces=exclude_subspaces, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 110, in find_packages_allowing_duplicates
    packages[path] = parse_package(os.path.join(basepath, path), warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/package.py", line 370, in parse_package
    return parse_package_string(f.read(), filename, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/package.py", line 402, in parse_package_string
    assert pkg.package_format in [1, 2], "Unable to handle package.xml format version '%d', please update catkin_pkg (e.g. on Ubuntu/Debian use: sudo apt-get update && sudo apt-get install --only-upgrade python-catkin-pkg)" % pkg.package_format
AssertionError: Unable to handle package.xml format version '3', please update catkin_pkg (e.g. on Ubuntu/Debian use: sudo apt-get update && sudo apt-get install --only-upgrade python-catkin-pkg)

ctucker@dawson:~/ros2_2018$ uname -a
Linux dawson 4.4.0-135-generic #161-Ubuntu SMP Mon Aug 27 10:45:01 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

ctucker@dawson:~/ros2_2018$ dpkg -l | grep python-catkin
ii  python-catkin-pkg                                                0.2.10-2                                         all          Low-level build system macros for Robot OS -- Python module

closed 2018-09-29 by ClarkTucker

m4xst3r, 2018-10-15:
Hi Clark,

how did you solve the issue? I'm facing the same Problem.

ClarkTucker, 2018-10-15:
Short answer: I started over with a clean install of Ubuntu 16.04.

I think that, some time in the past, I had manually installed some of the dependencies; and now, one or more of them were out of date.  I gave up trying to figure out what it was, and just started over.

m4xst3r, 2018-10-15:
Okay, Thank you! I'm going to do the same.

Infinity8sailor, 2020-06-04:
hey this will help solving issue
`sudo pip install -U rosdep rosinstall_generator wstool rosinstall`

uzgit, 2020-07-12:
@Infinity8sailor yes this worked for me.

dirk-thomas, 2020-07-13:
> `sudo pip install -U rosdep rosinstall_generator wstool rosinstall`

On a supported Ubuntu distro you should *not* install these packages via `pip` but using `apt` instead.

eric-schleicher, 2021-02-03:
I'm having this issue on a vanilla 18.04 system with the packages installed via apt, FWIW.    It's a jetson (Jetpack 4.5.1)
```
toaster@spacetoaster-006:~/workspaces/ros_catkin_ws$ dpkg -l | grep python-catkin
ii  python-catkin-pkg                             0.3.9-1                                          all          Low-level build system macros for Robot OS -- Python 2 module
```

it looks like the pip route 'works'; but always scared to let pip loose on ubuntu systems.   Is there a deliberate way to check the version that is being called by the ros build scripts?