ros-infrastructure/rosdep #635: os_detect issue

Henry-Wilson, 2018-10-19:
I am trying to build ros from source on OpenSUSE Leap 15. I am having an issue with the "rosdep update" command. Here it is:

```
wilsontux@linux-idrr:~> rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"

ERROR: Rosdep experienced an error: called in incorrect OS
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.12.2

Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3.6/site-packages/rosdep2/main.py", line 386, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python3.6/site-packages/rosdep2/main.py", line 395, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python3.6/site-packages/rosdep2/main.py", line 592, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python3.6/site-packages/rosdep2/sources_list.py", line 490, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python3.6/site-packages/rosdep2/gbpdistro_support.py", line 150, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python3.6/site-packages/rosdep2/__init__.py", line 87, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python3.6/site-packages/rosdep2/platforms/debian.py", line 53, in register_platforms
    register_linaro(context)
  File "/usr/lib/python3.6/site-packages/rosdep2/platforms/debian.py", line 70, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python3.6/site-packages/rosdep2/installers.py", line 115, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python3.6/site-packages/rospkg/os_detect.py", line 660, in get_name
    self.detect_os()
  File "/usr/lib/python3.6/site-packages/rospkg/os_detect.py", line 622, in detect_os
    self._os_codename = os_detector.get_codename()
  File "/usr/lib/python3.6/site-packages/rospkg/os_detect.py", line 241, in get_codename
    raise OsNotDetected('called in incorrect OS')
rospkg.os_detect.OsNotDetected: called in incorrect OS

wilsontux@linux-idrr:~>
```

I appreciate any help I can get. Thanks!

Henry-Wilson, 2018-10-20:
Alright, i tried --os "OpenSUSE" with the same result. Please advise further if there's more I can do. I'll take a look at the script, but I'm not an expert in python.

Henry-Wilson, 2018-10-20:
Okay I've looked at it more and it turns out that that particular detector is looking for files that do not exist. one is /etc/SuSE-brand, which is now /etc/SUSE-brand (blah) and another was /etc/SuSE-release, which I can't find an equivalent to anywhere on my system. I changed both file names to /etc/SUSE-brand in hopes it would find something at all to work with, but it still crashed on the same line.

Henry-Wilson, 2018-10-20:
I think I might be able to fix this with some help. Let me go put a ticket in on some openSUSE forums, see if I can find out where the equivalent file is these days.

Henry-Wilson, 2018-10-20:
Also, what it appears to be looking for is a CODENAME, which I can't find much about. I don't know what the codename of openSUSE 15 is.

Henry-Wilson, 2018-10-20:
So for os detection, this software should be updated to comply with the latest freedesktop specifications, where the file is /etc/os-release. 
This is the support thread for openSUSE.
https://forums.opensuse.org/showthread.php/533527-Old-files-what-are-the-new-locations?p=2883883#post2883883
I am going to try to get the script working for me, then send it up to you folks. I'll let you know how it goes.

Henry-Wilson, 2018-10-20:
Okay, I don't know how to fix this, but it looks like there is already an FDO implementation in there. Please help me figure out how to make my computer be detected with this method, as it is compliant to it.

Henry-Wilson, 2018-10-21:
Okay I fixed it. How should I give you guys the fix?

tfoote, 2018-10-22:
> Okay I fixed it. How should I give you guys the fix?
It sounds like SuSE has updated where it drops the marker files.  I'm glad to hear you got it working. A pull request on the appropriate repository would be the best way.

nuclearsandwich, 2018-11-05:
@Henry-Wilson it sounds like you have a patch to rospkg which has the openSUSE detector using FdoDetect. If you can open a PR for that change we can review it and potentially get it into the next rospkg release. https://github.com/ros-infrastructure/rospkg

Henry-Wilson, 2018-11-05:
Hey, Steven. I did make that patch, but it's not useful unless further 
issues are fixed. I ended up having to install Ubuntu on a different 
computer in order to use ROS. If you still want the patch, let me know 
and I'll see what I can do.

nuclearsandwich, 2018-11-05:
@Henry-Wilson if you could open a PR with the state of your patch and document what other issues are still outstanding it would be a great help to the next individual who wants to contribute improvements to the SUSE support.

gleichdick, 2019-10-15:
Hey folks, see also issue ros-infrastructure/rospkg#176. A PR (ros-infrastructure/rospkg#179) is already in progress